ros-naoqi/naoqi_driver #22: cannot receive image

k-okada, 2015-07-08:
I had following error message on pepper robot
```
front_camera Camera Handle is empty - cannot retrieve image
infrared_camera Camera Handle is empty - cannot retrieve image
front_camera Camera Handle is empty - cannot retrieve image
depth_camera Camera Handle is empty - cannot retrieve image
infrared_camera Camera Handle is empty - cannot retrieve image
front_camera Camera Handle is empty - cannot retrieve image
depth_camera Camera Handle is empty - cannot retrieve image
front_camera Camera Handle is empty - cannot retrieve image
infrared_camera Camera Handle is empty - cannot retrieve image
```

Karsten1987, 2015-07-10:
It's a known issue. For me this indicates a problem in naoqi ALVideoDevice.
This error message occurs directly when the proxy towards the ALVideoDevice doesn't deliver a correct image.

Rebooting naoqi or closing the cameras with the ALVideoDevice API may help here.

vrabaud, 2015-07-15:
a warning message explaining the status and what to do should be written then

Karsten1987, 2015-08-17:
fixed in 537716962e467985178b7f37cf6534cc0835d1ae

closed 2015-08-17 by Karsten1987