ros-naoqi/naoqi_driver #23: alrosbridge_bin does not stops on C-c

k-okada, 2015-07-08:
after pressing C-c, I had message as following:
```
^C[I] 5984 qi.Application: Sending the stop command...
[W] 5868 qitype.signal: Exception caught from signal subscriber: Call argument number 0 conversion failure from LogMessage to LogMessage. Function signature: (LogMessage).
[W] 5868 qitype.signal: Exception caught from signal subscriber: Call argument number 0 conversion failure from LogMessage to LogMessage. Function signature: (LogMessage).
[W] 5867 qitype.signal: Exception caught from signal subscriber: Call argument number 0 conversion failure from LogMessage to LogMessage. Function signature: (LogMessage).
[W] 5868 qitype.signal: Exception caught from signal subscriber: Call argument number 0 conversion failure from LogMessage to LogMessage. Function signature: (LogMessage).
[W] 5866 qitype.signal: Exception caught from signal subscriber: Call argument number 0 conversion failure from LogMessage to LogMessage. Function signature: (LogMessage).
```

Karsten1987, 2015-07-10:
Can you reproduce this all times? How do you launch the bridge?
I am not sure if this might not be an naoqi related issue. 

For me C-c shuts down nicely.
```
^C[I] 1044 qi.Application: Sending the stop command...
Audio Extractor: Stop
Unsubscribe camera handle front_camera_0
ALRosBridge is shutting down..
```

k-okada, 2015-07-11:
>
> Can you reproduce this all times? How do you launch the bridge?
> I am not sure if this might not be an naoqi related issue.

yes, and i ran like this
 ./devel/lib/naoqi_rosbridge/alrosbridge_bin
--qi-url=tcp://xxx.xx.xx.xxx.:9559 xxx.xxx.xxx.xxx.

I thought both qi and ros has signal handler and could not catch C-c well,
but if this is working on your side, may be something wrong on my side...

 For me C-c shuts down nicely.

Karsten1987, 2015-07-15:
That is right, both have signal handlers. For this case, we disabled the ROS sigint for letting qi trigger the shutdown process.
https://github.com/ros-naoqi/alrosbridge/blob/master/src/ros_env.hpp#L84

Karsten1987, 2015-08-17:
does this still occur with the latest updates?

k-okada, 2015-10-02:
Thanks this is fixed now.

closed 2015-10-02 by k-okada