ros-naoqi/naoqi_driver #33: Bus Error

zazics, 2015-08-18:
Hello,

I've built "naoqi_driver" on a Raspberry Pi 2 using "catkin_make_isolated". Except this warning "extended initializer lists only available with std=c++0x or -std=gnu++0x", everything built correctly.
But, when I run:
"rosrun naoqi_driver naoqi_driver_node --qi-url=<robot-ip> wlan0"
I get a "Bus error" message and process stops. 

I have no problem using libqi in another C++ program to control Nao from my Raspberry Pi.

Karsten1987, 2015-08-18:
please verify that you start the bridge correctly and that you are on the latest version.
You can find the instructions on how to start your node here:
http://ros-naoqi.github.io/naoqi_driver/start.html#start-the-naoqi-driver-module

zazics, 2015-08-20:
Yes, I downloaded the latest version a few days ago. I followed that documentation. If I undertand it well, I only have to start "roscore" and "rosrun naoqi_driver ....". Am I right ?
Maybe my problem is due to ARM architecture of Raspberry ?

vrabaud, 2015-08-21:
You get a memory corruption right ? (please write your error message again).
We've never tested that bridge or libqi with raspberry pi so that could indeed be a problem.
@Karsten1987 , maybe you can show him how to disable certain modules so that we narrow down the corruption.

zazics, 2015-08-21:
my error message is just "Bus error". Nothing more, nothing less.
"Libqi" seems to work well on a Raspberry Pi 2. I can send order (move, speech, joint) to Nao,  read memory and react to events.
I found this thread: https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=102400
It talks about some hard float point / memory alignment problem.
I will try to investigate more as soon as possible.