ros-naoqi/naoqi_driver #40: when requesting VGA on depth sensor, getting only QQQVGA (80x60)

laurent-george, 2015-09-14:
rostopic echo /pepper_robot/camera/depth/image_raw --noarr 

header: 
  seq: 76
  stamp: 
    secs: 1442246921
    nsecs: 366206631
  frame_id: CameraDepth_optical_frame
height: 60
width: 80
encoding: 16UC1
is_bigendian: 0
step: 160

in my boot_config.json

    "depth_camera":
    {
      "enabled"       : true,
      "resolution"    : 2,
      "fps"           : 10,
      "recorder_fps"  : 10
    },

everything works fine for resolution 1 (320x240)

vrabaud, 2015-09-16:
Not supported on Pepper. A warning should probably be logged though.

nlyubova, 2016-04-19:
@vrabaud do you know why it is not supported on Pepper? I can get VGA after few modifications in soft but I get low fps. Do you know what was the issue ?

vrabaud, 2016-04-19:
That is to see with Corentin.

nlyubova, 2017-04-06:
It is true @laurent-george  and it comes from Naoqi and not from Naoqi Driver. 
Seams impossible to solve, and the only possibility is to use QVGA or smaller

mikaelarguedas, 2017-04-06:
I think that the warning suggested in https://github.com/ros-naoqi/naoqi_driver/issues/40#issuecomment-140737015 should be added to avoid confusion

nlyubova, 2017-04-06:
@mikaelarguedas right, I never missed this comment

warp1337, 2017-11-15:
When you look at the specs of the camera it indeed is capable of VGA. Can this be fixed in the NaoQi API?

nlyubova, 2017-11-21:
The VGA is of worse quality than QVGA and this limitation comes from Naoqi but not from Naoqi Driver. Please, use QVGA.

nlyubova, 2017-11-21:
@mikaelarguedas : please check https://github.com/ros-naoqi/naoqi_driver/pull/93

closed 2017-11-21 by nlyubova