ros-naoqi/naoqi_driver #50: No distortion_model and D param in /pepper_robot/camera/depth/camera_info

yoshimalucky, 2015-10-19:
The values for distortion_model and D is blank in /pepper_robot/camera/depth/camera_info like below:

distortion_model: plumb_bob
D: [0.0, 0.0, 0.0, 0.0, 0.0]

Is it difficult to publish these values?
Because of this, I can not use depthimage_to_laserscan.

Karsten1987, 2015-10-19:
Hi,

I am not sure why, but exactly these two lines for the distortion model are commented in https://github.com/ros-naoqi/naoqi_driver/blob/master/src/converters/camera_info_definitions.hpp#L176
If you can, just uncomment them, recompile and check if you succeed in transforming into a laserscan. 

This can definitely be a PR.

yoshimalucky, 2015-10-21:
Thank you so much! I'll check it out and report how it works.

vrabaud, 2016-01-01:
@yoshimalucky , any news on that ? thx.

dac31415, 2021-09-24:
After I uncomment and tried to recompile, the following mistake arises. 
/opt/ros/melodic/include/qi/detail/future_fwd.hpp:750:5: note: declared here
     operator const typename Future<T>::ValueTypeCast&() const          { _sync = false; return _future.value(); }
     ^~~~~~~~
naoqi_driver/CMakeFiles/naoqi_driver.dir/build.make:230: recipe for target 'naoqi_driver/CMakeFiles/naoqi_driver.dir/src/converters/camera.cpp.o' failed
make[2]: *** [naoqi_driver/CMakeFiles/naoqi_driver.dir/src/converters/camera.cpp.o] Error 1
CMakeFiles/Makefile2:9042: recipe for target 'naoqi_driver/CMakeFiles/naoqi_driver.dir/all' failed
make[1]: *** [naoqi_driver/CMakeFiles/naoqi_driver.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Any reasons why?

mbusy, 2021-09-24:
My guess would be that it's a boost related issue. Take a look at this version of the file [here](https://github.com/softbankrobotics-research/naoqi_driver/blob/master/src/converters/camera_info_definitions.hpp) (softbank robotics research's fork), this should allow you to get rid of boost for that file, and include the distortion model. What we use in the file is pretty old (for retro-compatibility purposes), you could consider using the functionalities of the standard lib instead.

dac31415, 2021-09-24:
Thanks again. @mbusy I did change it. No error but the issue is still the same.

dac31415, 2021-09-27:
@mbusy the problem might be when with the  pepper bringup package.
I executed the driver:
`roslaunch` naoqi_driver naoqi_driver.launch naop_ip:=<nao_ip> network_interface:=<networ...>
after that I can view (either with rviz or rqt_image_view) all the raw images from the cameras with the topics
`/naoqi_driver/camera/.../image_raw`
and also execute teleop with `roslaunch` nao_teleop teleop_joy.launch

mbusy, 2021-09-27:
Thanks @dac31415. We probably won't be investigating the errors in the pepper bringup package, we want to focus the maintaining efforts on the driver. That being said, PRs would definitely be considered / reviewed / merged

dac31415, 2021-09-28:
Ok, thanks @mbusy. I'll keep looking on how to solve it. The naoqi_driver is working fine for image viewing but is when I try to convert the data from the cameras. (depth to point cloud) as I haven't been able to do it. So I don't know if it's the pepper_bringup or I am missing some other dependencies or packages.

UdayRockzz, 2021-11-26:
Hi @dac31415 , Did you solve this issue(depth to point cloud)?

dac31415, 2021-11-26:
@UdayRockzz follow steps from @mbusy (softbank robotics research's fork)

UdayRockzz, 2021-11-28:
@dac31415 Thanks a lot for responding. I have followed all those steps which @mbusy suggested but still, I am unable to get the point cloud information. If it is possible could you please share the file with me? It will be helpful for my analysis.

redsIng, 2023-04-13:
I have exactly the same problem. In addition I have uncommented those lines in camer_info_definitions.hpp and corrected the resulting error adding ".convert_to_container<std::vector<double> >();" to all cam_info_msg.D lines like how it has been done for front and bottom camera in the same file. I compile it and all seems work. When I do depth camera calibration following [Pepper/Tutorial/Calibration](http://wiki.ros.org/pepper/Tutorials/Calibration) guide to generate .yaml file and then use [depthimage_to_laserscan](http://wiki.ros.org/depthimage_to_laserscan) ros package, the front and bottom camera works but the depth camera can't calibrate so I can't use depthimage_to_laserscan. Can you provide me any suggestion on how fix it? Am I doing it wrong?
Thanks in advance.

This is my camera_info_definitions.hpp without the correction that I did:
`
#ifndef CAMERA_INFO_DEF_HPP
#define CAMERA_INFO_DEF_HPP

#include <sensor_msgs/CameraInfo.h>
#include <boost/assign/list_of.hpp>

namespace naoqi
{
namespace converter
{
namespace camera_info_definitions
{

/**
* TOP CAMERA
*/
inline sensor_msgs::CameraInfo createCameraInfoTOPVGA()
{
  sensor_msgs::CameraInfo cam_info_msg;

  cam_info_msg.header.frame_id = "CameraTop_optical_frame";

  cam_info_msg.width = 640;
  cam_info_msg.height = 480;
  cam_info_msg.K = boost::array<double, 9>{{ 556.845054830986, 0, 309.366895338178, 0, 555.898679730161, 230.592233628776, 0, 0, 1 }};

  cam_info_msg.distortion_model = "plumb_bob";
  cam_info_msg.D = boost::assign::list_of(-0.0545211535376379)(0.0691973423510287)(-0.00241094929163055)(-0.00112245009306511)(0).convert_to_container<std::vector<double> >();

  cam_info_msg.R = boost::array<double, 9>{{ 1, 0, 0, 0, 1, 0, 0, 0, 1 }};

  cam_info_msg.P = boost::array<double, 12>{{ 551.589721679688, 0, 308.271132841983, 0, 0, 550.291320800781, 229.20143668168, 0, 0, 0, 1, 0 }};

  return cam_info_msg;
}

inline sensor_msgs::CameraInfo createCameraInfoTOPQVGA()
{
  sensor_msgs::CameraInfo cam_info_msg;

  cam_info_msg.header.frame_id = "CameraTop_optical_frame";

  cam_info_msg.width = 320;
  cam_info_msg.height = 240;
  cam_info_msg.K = boost::array<double, 9>{{ 274.139508945831, 0, 141.184472810944, 0, 275.741846757374, 106.693773654172, 0, 0, 1 }};

  cam_info_msg.distortion_model = "plumb_bob";
  cam_info_msg.D = boost::assign::list_of(-0.0870160932911717)(0.128210165050533)(0.003379500659424)(-0.00106205540818586)(0).convert_to_container<std::vector<double> >();

  cam_info_msg.R = boost::array<double, 9>{{ 1, 0, 0, 0, 1, 0, 0, 0, 1 }};

  cam_info_msg.P = boost::array<double, 12>{{ 272.423675537109, 0, 141.131930791285, 0, 0, 273.515747070312, 107.391746054313, 0, 0, 0, 1, 0 }};

  return cam_info_msg;
}

inline sensor_msgs::CameraInfo createCameraInfoTOPQQVGA()
{
  sensor_msgs::CameraInfo cam_info_msg;

  cam_info_msg.header.frame_id = "CameraTop_optical_frame";

  cam_info_msg.width = 160;
  cam_info_msg.height = 120;
  cam_info_msg.K = boost::array<double, 9>{{ 139.424539568966, 0, 76.9073669920582, 0, 139.25542782325, 59.5554242026743, 0, 0, 1 }};
[... 222 lines cut ...]
{
    return createCameraInfoStereo(1280, 360, 2.0);
}

inline sensor_msgs::CameraInfo createCameraInfoStereoQQ720PX2()
{
    return createCameraInfoStereo(640, 180, 4.0);
}

inline sensor_msgs::CameraInfo createCameraInfoStereoQQQ720PX2()
{
    return createCameraInfoStereo(320, 90, 8.0);
}

inline sensor_msgs::CameraInfo createCameraInfoStereoQQQQ720PX2()
{
    return createCameraInfoStereo(160, 45, 16.0);
}

} // camera_info_definitions
} //publisher
} //naoqi

#endif
`