ros-naoqi/naoqi_driver #54: running naoq drover on choregraphe

k-okada, 2015-11-09:
naoqi_driver checks `RobotConfig/Body/Type` but it seems this is not work on Choregraphe 2.1.4.
Is this is issue of choregraphe and this will fixed anytime soon?
or if we have to check in naoqi_driver side, is there anyway to know robot type when we connect to choreographe? or if not is there way to find if we connect to realrobot or virtual robot?
```
  // Get the robot type
  std::cout << "Receiving information about robot model" << std::endl;
  qi::AnyObject p_memory = session->service("ALMemory");
  std::string robot = p_memory.call<std::string>("getData", "RobotConfig/Body/Type" );
  std::transform(robot.begin(), robot.end(), robot.begin(), ::tolower);
```
https://github.com/ros-naoqi/naoqi_driver/blob/master/src/helpers/driver_helpers.cpp#L44

suryaambrose, 2016-02-03:
Hi, 

I don't exactly know the reason of the problem you face, but I think that key only exist on real robots. I don't have it either on more recent choregraphe. I don't really know if it is a bug or an expected behavior. 

Anyway, you can use the key "Dialog/RobotModel" instead. I don't know if it is exactly the same behavior as RobotConfig/Body/Type, but at least the information is there and existing.

warp1337, 2017-03-22:
Bump. I have the same error when trying to run a virtual Pepper via Chore or naoqi-bin

<pre>
<code>
[14:44]$ roslaunch naoqi_driver naoqi_driver.launch nao_ip:=thorium.local roscore_ip:=thorium network_interface:=eth0
... logging to /home/fl/.ros/log/4eaec0ec-0f03-11e7-9827-9890969b8671/roslaunch-thorium-17951.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://thorium:45905/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[naoqi_driver-1]: started with pid [17969]
Receiving information about robot model
terminate called after throwing an instance of 'qi::FutureUserException'
  what():       ALMemory::getData
        Data not found RobotConfig/Body/Type
<code>
</pre>

nlyubova, 2017-03-22:
Which version of Naoqi installed on your robot?

warp1337, 2017-03-23:
We dont have our Pepper Robot yet (Bielefeld University) it about to be delivered this week hopefully. I am currently implementing a Simulation for the Pepper as an alternative to gazebo. Thus I am starting a naoqi-bin and connect to it. I think that this Memory is not available in naoqi without the actual hardware?

nlyubova, 2017-11-21:
There is actually a possibility to run NAOqi on your PC (download SDK, go to its path and run "naoqi --broker-ip 127.0.0.1") and then run Naoqi Driver. So, you do not need a Choregraphe

warp1337, 2017-11-30:
Hey, I have a solution for this. I got the driver running with **Choregraphe** by inserting the missing mem keys manually *can share the code). However, I found out, that the driver does NOT run using simulated time. 

<pre>
<param name="/use_sim_time" value="true"/>
</pre>

Does anyone know why this is? This is a must have in order to use the driver for simulation.

farhangnaderi, 2020-05-19:
@warp1337 Can you share your code? Am getting same error.

ysys98, 2022-03-09:
Hello, have you solved this problem? I have the same problem

farhangnaderi, 2022-03-10:
@ysys98 I gave up. The whole setup procedure is so outdated and not a match with my system anymore. So it was not worth my time.