ros-naoqi/naoqi_driver #57: bottom_cmaeraCmaera Handle is empty, and can not see front camer on rostopic, but we can see front camera on choregraph

k-okada, 2016-01-08:
```
bottom_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
bottom_cameraCamera Handle is empty - cannot retrieve image
bottom_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
bottom_cameraCamera Handle is empty - cannot retrieve image
bottom_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
bottom_cameraCamera Handle is empty - cannot retrieve image
bottom_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
bottom_cameraCamera Handle is empty - cannot retrieve image
bottom_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
^Cbottom_cameraCamera Handle is empty - cannot retrieve image
bottom_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
``

Karsten1987, 2016-01-08:
I was struggling with this error as well. I haven't seen a way to fix that on any ROS side. It seems to me that this situation occurs when there are zombie subscriber to the ALVideoDevice, which finally exceed the maximal number of subscribers.
@k-okada Can you somehow reproduce this behavior regularly? If so, please check the number of subscribers.

suryaambrose, 2016-01-08:
@Karsten1987 you are right :).
This happens when you launch "something" that subscribes to a camera but does not unsubscribe. Launch this script 6 or 7 times and then this error happens. 

If you can reproduce this issue often, you are probably subscribing to a camera without unsubscribing afterwards.

k-okada, 2016-01-26:
OK, I found that it does happens if we enable audio, which fail to call "unsubscribe" at https://github.com/ros-naoqi/naoqi_driver/blob/master/src/event/audio.cpp#L115 . Or to be more precise, it calls "unsubscribe" but never returns, whcn we type "C-c". Do you have any idea?
```
index a9e0bd3..e4182cd 100644
--- a/share/boot_config.json
+++ b/share/boot_config.json
@@ -72,7 +72,7 @@
     },
     "audio":
     {
-      "enabled"       : true
+      "enabled"       : false
     },
     "bumper":
     {
```

k-okada, 2016-01-26:
should we call `subscribe` before `setClientPreferences` ? https://github.com/ros-naoqi/naoqi_driver/blob/master/src/event/audio.cpp#L95

suryaambrose, 2016-01-26:
Seriously  O_o ?? I really don't see why audio would cause a camera issue. Can you provide a protocol to reproduce this issue please ?

For your second question, it is better to call setClientPreferences before subscribing so that is not the issue.

k-okada, 2016-01-26:
if we enable audio, C-c did not work, so the ros process did not shutdown
cleanly, that'd did not unsubscribe the camera correctly, if we disable
audio, C-c works.

suryaambrose, 2016-01-26:
Ok I understand. Maybe we should make another issue then ? This needs further investigation.

k-okada, 2016-01-26:
sure, please confirm current master did not stop cleanly and create PR.

PR #62 (2016-02-03): Fix termination issues
This PR fixes the segfault happening every time the bridge is terminated.

@k-okada I also took the liberty of including your commit to fix the termination error due to tactile and bumpers, hope you don't mind. If you do I'll remove it. 

This fixes #57 and the other problems discussed in #61 (not including the deadlock caused by audio yet, but it is on the way)
closed 2016-02-03 by k-okada