ros-naoqi/naoqi_driver #66: I cannot compile naoqi_driver with ros-indigo-naoqi-libqi version 2.5

kochigami, 2016-03-16:
Hi,
I'd like to ask someone for my question. Could anyone help me?
I am very sorry for my trouble.

I cannot compile ```naoqi_driver``` from this page (branch master) with ```ros-indigo-naoqi-libqi 2.5.0-1trusty-20160313-144036-0700```.
I use ubuntu 14.04, ros indigo.

You will find the all compile log below, and the error seems these messages. 
```
/opt/ros/indigo/include/qi/type/typeinterface.hpp:396:51: error: QI_TYPE_ENUM_REGISTER_ is deprecated (declared at /opt/ros/indigo/include/qi/type/typeinterface.hpp:387) [-Werror=deprecated-declarations]
       static const detail::QI_TYPE_ENUM_REGISTER_ BLAH;              \
                                                   ^
/home/kochigami/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:1: note: in expansion of macro QI_TYPE_ENUM_REGISTER
 QI_TYPE_ENUM_REGISTER(naoqi::Topics);
```
I tried to downgrade ```ros-indigo-naoqi-libqi``` , but ```sudo apt-cache showpkg ros-indigo-naoqi-libqi``` says that I can install only version 2.5 . 

1. If the result of cat ``` /etc/apt/sources.list.d/ros-latest.list ``` is ``` deb http://packages.ros.org/ros-shadow-fixed/ubuntu trusty main ```

```
Package: ros-indigo-naoqi-libqi
Versions: 
2.5.0-1trusty-20160313-144036-0700 (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/dpkg/status
                  MD5: 11916c51ef94db9fc8a3aac342be977d

Reverse Depends: 
  ros-indigo-naoqi-driver,ros-indigo-naoqi-libqi
  ros-indigo-naoqi-libqicore,ros-indigo-naoqi-libqi
Dependencies: 
2.5.0-1trusty-20160313-144036-0700 - libboost-chrono1.54.0 (0 (null)) libboost-filesystem1.54.0 (0 (null)) libboost-program-options1.54.0 (0 (null)) libboost-regex1.54.0 (0 (null)) libboost-system1.54.0 (0 (null)) libboost-thread1.54.0 (0 (null)) libc6 (2 2.15) libgcc1 (2 1:4.1.1) libstdc++6 (2 4.6) libboost-all-dev (0 (null)) 
Provides: 
2.5.0-1trusty-20160313-144036-0700 - 
Reverse Provides: 
```

2. If the result of cat ``` /etc/apt/sources.list.d/ros-latest.list ``` is ``` deb http://packages.ros.org/ros/ubuntu trusty main ```

```
Package: ros-indigo-naoqi-libqi
Versions: 
2.5.0-1trusty-20160313-144036-0700 (/var/lib/apt/lists/packages.ros.org_ros-shadow-fixed_ubuntu_dists_trusty_main_binary-amd64_Packages) (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/packages.ros.org_ros-shadow-fixed_ubuntu_dists_trusty_main_binary-amd64_Packages
                  MD5: 11916c51ef94db9fc8a3aac342be977d

Reverse Depends: 
  ros-indigo-naoqi-driver,ros-indigo-naoqi-libqi
  ros-indigo-naoqi-libqi:i386,ros-indigo-naoqi-libqi
  ros-indigo-naoqi-libqicore,ros-indigo-naoqi-libqi
Dependencies: 
2.5.0-1trusty-20160313-144036-0700 - libboost-chrono1.54.0 (0 (null)) libboost-filesystem1.54.0 (0 (null)) libboost-program-options1.54.0 (0 (null)) libboost-regex1.54.0 (0 (null)) libboost-system1.54.0 (0 (null)) libboost-thread1.54.0 (0 (null)) libc6 (2 2.15) libgcc1 (2 1:4.1.1) libstdc++6 (2 4.6) libboost-all-dev (0 (null)) ros-indigo-naoqi-libqi:i386 (0 (null)) 
Provides: 
2.5.0-1trusty-20160313-144036-0700 - 
Reverse Provides: 
```

This is the log when I tried to compile naoqi_driver by ```catkin clean -a``` and ```catkin build```
```
cat /home/kochigami/catkin_ws/build/build_logs/naoqi_driver.log
[naoqi_driver] ==> '/home/kochigami/catkin_ws/build/naoqi_driver/build_env.sh /usr/bin/cmake /home/kochigami/catkin_ws/src/naoqi_driver -DCATKIN_DEVEL_PREFIX=/home/kochigami/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/kochigami/catkin_ws/install' in '/home/kochigami/catkin_ws/build/naoqi_driver'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
[... 152 lines cut ...]
                 from /opt/ros/indigo/include/qi/signal.hpp:12,
                 from /opt/ros/indigo/include/qi/anyobject.hpp:15,
                 from /home/kochigami/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:30,
                 from /home/kochigami/catkin_ws/src/naoqi_driver/include/naoqi_driver/recorder/globalrecorder.hpp:24,
                 from /home/kochigami/catkin_ws/src/naoqi_driver/src/recorder/globalrecorder.cpp:21:
/opt/ros/indigo/include/qi/type/typeinterface.hpp:396:51: error: QI_TYPE_ENUM_REGISTER_ is deprecated (declared at /opt/ros/indigo/include/qi/type/typeinterface.hpp:387) [-Werror=deprecated-declarations]
       static const detail::QI_TYPE_ENUM_REGISTER_ BLAH;              \
                                                   ^
/home/kochigami/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:1: note: in expansion of macro QI_TYPE_ENUM_REGISTER
 QI_TYPE_ENUM_REGISTER(naoqi::Topics);
 ^
/opt/ros/indigo/include/qi/type/typeinterface.hpp:396:51: error: QI_TYPE_ENUM_REGISTER_ is deprecated (declared at /opt/ros/indigo/include/qi/type/typeinterface.hpp:387) [-Werror=deprecated-declarations]
       static const detail::QI_TYPE_ENUM_REGISTER_ BLAH;              \
                                                   ^
/home/kochigami/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:66:1: note: in expansion of macro QI_TYPE_ENUM_REGISTER
 QI_TYPE_ENUM_REGISTER(naoqi::dataType::DataType);
 ^
cc1plus: some warnings being treated as errors
make[2]: *** [CMakeFiles/naoqi_driver.dir/src/recorder/globalrecorder.cpp.o]  1
cc1plus: some warnings being treated as errors
make[2]: *** [CMakeFiles/naoqi_driver.dir/src/naoqi_driver.cpp.o]  1
make[1]: *** [CMakeFiles/naoqi_driver.dir/all]  2
make: *** [all]  2
[naoqi_driver] <== '/home/kochigami/catkin_ws/build/naoqi_driver/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2
```

vrabaud, 2016-03-16:
hi, sorry about that, we are waiting for a fix from @suryaambrose . I cannot help on that for the next two weeks. @suryaambrose , should we revert ?
Also, @kochigami , can you please try removing that line and see if it work for you: https://github.com/ros-naoqi/naoqi_driver/blob/master/CMakeLists_catkin.txt#L30
Thx.

suryaambrose, 2016-03-16:
@kochigami  yes the problem is known. The line giving you an error is easy to fix, but you will then run into boost incompatibilities issues. 
@vrabaud give me until this week-end. If I can't figure out a solution, then yes we will revert.

kochigami, 2016-03-16:
@vrabaud @suryaambrose 
Thank you for your advice.
 I changed CMakeLists.txt, tools.hpp and external_registration.cpp as same as #64 
and deleted the line in CMakeLists_catkin.txt. 
Then I found boost imcompatible errors. 

```
/opt/ros/indigo/lib/libqi.so: `boost::filesystem::detail::copy_file(boost::filesystem::path const&, boost::filesystem::path const&, boost::filesystem::copy_option, boost::system::error_code*)' 
collect2: error: ld returned 1 exit status
```

Thank you very much for your help in your busy time.

vrabaud, 2016-03-16:
oh, totally forgot about #64 : @suryaambrose , feel free to merge if that is a fix. And also remove https://github.com/ros-naoqi/naoqi_driver/blob/master/CMakeLists_catkin.txt#L30 if that works for you. Thx.

suryaambrose, 2016-03-16:
#64 fixes compilation, but not the linkage towards boost. Let's not merge it right now (less things to revert if we decide to go back to libqi-2.3)

suryaambrose, 2016-03-19:
This issue is more about libqi than naoqi_driver. 
I reported everything needed here: https://github.com/ros-naoqi/libqi-release/issues/3

closed 2016-03-19 by suryaambrose