ros-naoqi/naoqi_driver #68: LogMessage warning

nlyubova, 2016-04-15:
Do you know how to fix this warning? when running the most recent naoqi 

[W] 4822 qitype.signal: Exception caught from signal subscriber: Call argument number 0 conversion failure from LogMessage to LogMessage. Function signature: (LogMessage).

severin-lemaignan, 2017-03-17:
Any news on that issue? my console is literally flooded with these messages...

suryaambrose, 2017-03-18:
Seems like a libqi warning. What is the naoqi version ? Is the naoqi_driver you use the released package or a custom compiled one ? If compiled, with which compiler (qibuild or catkin), which ROS distro and which libqi version ?

nlyubova, 2017-03-18:
Naoqi 2.5, both release and source compiled with catkin. Both ROS Indigo and Kinetic

warp1337, 2017-04-02:
Same here. Still no fix?

suryaambrose, 2017-04-03:
-_- ok this is most likely due to an API change between libqicore 2.3 (the only libqicore released in Indigo and Kinetic) and libqicore 2.5 (in the robot). The LogMessage structure is different. 

Upgrading the ros-**-libqicore package to libqicore 2.5 seems the easiest solution, but then the warning will appear for people using naoqi 2.3... 
A good compromise might be to upgrade libqicore only on Kinetic. Users of naoqi 2.3 would have to use Indigo (an old ROS distro for an old naoqi), and users of naoqi 2.5 would have to use Kinetic.

What do you think ?

nlyubova, 2017-04-03:
but how can we fix it on Indigo? The problem is exactly the same for Indigo and Kinetic.
Let's move  libqicore to 2.5 for both indigo and Kinetic, can we?

suryaambrose, 2017-04-03:
Which mean people using naoqi 2.3 will have the issue instead, and no way to fix it. I don't think we can do this, at least as long as naoqi 2.5 is not officially released (it is still in beta for now)

mtlazaro, 2017-04-28:
Same issue here in a fresh installation of Ubuntu 16.04 + ROS Kinetic and a recently acquired Pepper with naoqi version 2.5.3.3. naoqi_driver compiled from source with catkin.

warp1337, 2017-04-28:
You could try to upgrade your Pepper's OS to the latest version which available via Softbank's web interface for your robot. However, I didn't test it and it's probably still alpha/beta status. If you do so, you may provide some feedback how it went?

mtlazaro, 2017-05-03:
Hi,
thanks for your reply.

We have Pepper with (what we think is) the last version. Actually when we
started it for the first time it automatically updated all the software.
Version is 2.5.3.3.
If there is some other later version our Pepper is not being updated
automatically to that one.

Best,
Mayte.

awesomebytes, 2017-05-04:
Any workaround?

dgandiaga, 2017-05-30:
Hi guys!
How is this going? I have the same problem with naoqi 2.5 and I really need to use that version for the new functionalities about the charging base. Is there any fix for any ROS distribution? I'm using Jade, but I can change it if necessary.

warp1337, 2017-05-31:
I guess no. Our workaround is the python version (naoqi_driver_py).

dgandiaga, 2017-05-31:
Thanks for your answer. I thought the C++ version was the most advanced one. So you are focusing in the Python version? Doesn't it lack some of the sensors the C++ version already has integrated? Should I change to the Python one?

warp1337, 2017-05-31:
We are using the Python version, works good. It supports cameras [rgb and depth], lasers, IMU (not so sure), the ALNav package and joint control (not DCM).

dgandiaga, 2017-05-31:
Which of the versions has a better acquisition frequency? Because I am having a problem with that, the laser and the camera send the data at 0.5 Hertz, more or less (reading comments I have seen it may be a problem of my router, I'll switch it for another better).
The Python version doesn't support the microphone? That's a problem for my application, the C++ supports it (or at least it emits something in the /pepper_robot/naoqi_driver/audio topic, I still haven't been capable of playing this audio stream in my computer).

warp1337, 2017-06-01:
@dgandiaga Well ;) as I said the implementation works good. However, the initial implementation is not very efficient, since it queries ALMemory basically for every ray in every laser device. I fixed this in my fork using getListData. Here's the most important change (see: self.memProxy.getListData(tmp_array)) with this you will be able to obtain 6.66 Hz which is the maximum frequency available with Peppers' hardware.

<pre>
def fetchLaserValues(self, keyPrefix, scanNum):
        ranges = []
        # traverse backwards
        tmp_array = []
        for i in xrange(scanNum, 0, -1):
            keyX = keyPrefix + 'Seg' + '%02d' % (i,) + '/X/Sensor/Value'
            keyY = keyPrefix + 'Seg' + '%02d' % (i,) + '/Y/Sensor/Value'
            tmp_array.append(keyX)
            tmp_array.append(keyY)
        memData = self.memProxy.getListData(tmp_array)
        for i in xrange(0, len(tmp_array), 2):
            x = memData[i]
            y = memData[i+1]
            ranges.append(math.sqrt(math.pow(x, 2.0) + math.pow(y, 2.0)))
        return ranges
<pre>

Since this is a little off-topic considering the CPP node. Drop me a line via mail for more info. I will merge these kind of fixes after the RoboCup 2017.

nlyubova, 2017-06-01:
Hello @warp1337 could you do a pull request?

warp1337, 2017-06-01:
@nlyubova @dgandiaga Done. 

https://github.com/ros-naoqi/pepper_robot/pull/43

awesomebytes, 2017-07-31:
A workaround to avoid this annoying and CPU consuming continuous warning is to change the share/boot_config.json to disable the logs. Sorry I'm on a phone so I can't really link to it.