ros-naoqi/naoqi_driver #69: [question] Where to write dynamic reconfigure for speech recognition?

kochigami, 2016-05-24:
Hi, 
I'd like to include speech recognition handler in naoqi_driver.
reason:
```nao_apps/speech.launch``` has speech recognition service.
However, if I launch this with ```naoqi_driver```, two ```/speech``` topics from ```nao_apps``` and ```naoqi_driver``` react.   Pepper speaks twice.

I wrote two services ```/start_speech_recognition```, ```/stop_speech_recognition```, but I couldn't know where to write dynamic reconfigure server to set parameters before calling ```/start_speech_recognition```.
Could anyone give me advice?

After I learn the right place to write dynamic reconfigure server, I'll continue my work by following this tutorial:
http://wiki.ros.org/dynamic_reconfigure/Tutorials/SettingUpDynamicReconfigureForANode%28cpp%29 

What I did is below:
https://github.com/ros-naoqi/naoqi_driver/compare/master...kochigami:add-speech-recognition

k-okada, 2016-06-06:
> master...kochigami:add-speech-recognition

created pr at #70,

kochigami, 2016-06-09:
Thank you very much for your help.
I forgot to see the discussion in #41, and raised this issue. Sorry for that.

closed 2016-06-09 by kochigami