ros-naoqi/naoqi_driver #73: empty string can be passed to ros::init as node name

furushchev, 2016-09-11:
In the current code, ros node is initialized at https://github.com/ros-naoqi/naoqi_driver/blob/master/src/ros_env.hpp#L93.
In this line, when this node is launched by `roslaunch`, `ROS_NAMESPACE` is set on environment variable, so `ns_env` is not `NULL`, so node name which is passed to `ros::init` is `::naoqi::ros_env::getPrefix()`, but without passing argument `--namespace=`, `::naoqi::ros_env::getPrefix()` returns empty string (https://github.com/ros-naoqi/naoqi_driver/blob/master/src/external_registration.cpp#L73), which means ros node is initialized with empty node name.
Since, initializing ros node with empty string is totally invalid (see https://github.com/ros/ros_comm/issues/891), so we must initialize ros with some node name.

In current code, also arguments passed from ROS tools (e.g. `roslaunch`) are totally removed before passing them to `ros::init` by `ros::removeROSArgs` (https://github.com/ros-naoqi/naoqi_driver/blob/master/src/external_registration.cpp#L34, (I really cannot understand why it is necessary)), so remapping node name by `roslaunch` does also not work at all.

PacoDu, 2017-02-08:
Hi, i'm facing this issue when launching naoqi for Pepper robot. naoqi-driver thread is crashing with this error : 
```
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  The node name must not be empty
```

Did you manage to overcome this error ?

Karsten1987, 2017-02-08:
From what I understand, this is looks like a clear flaw to me. Do you mind opening a pull request?
The latest change in this regard, was made here:
https://github.com/ros-naoqi/naoqi_driver/commit/d325cda400cb1324c63c7ce19398c0e4f8809ec8

Also, the reasoning for all this complex behavior of removing ros args etc. comes due to the fact that the initial code ought to be executed completely ROS agnostic within a libqi framework.