ros-naoqi/naoqi_driver #77: Depth Image Encoding 16UC1 => mono16

hwp, 2016-12-14:
The depth image encoding in naoqi_driver is `16UC1` rather than `mono16`. Is there any reason to do this? 
As I understand the cv_bridge does not recognize `16UC1`. From time to time, I get error messages like:

```
[ERROR]: Unable to convert 16UC1 image to bgr8
```

suryaambrose, 2016-12-14:
I don't think there is. Does it cause any problem for you ? If it does, would you mind trying to change 16UC1 to mono16 and see if it works fine ?
I know your error message can appear when an image_view node receives a 16UC1 image. Maybe other nodes can print that same message.

hwp, 2016-12-14:
Indeed I was trying to use image_saver from image_view to convert images in rosbag to png files. and it has the error.
Once I changed 16UC1 to mono16 (by editing the rosbag file), it works.

hwp, 2016-12-14:
And, there is also an issue when I try to rectify the depth image by image_proc. the error I get is:

```
[ERROR] [1481728189.791698687]: Raw image topic '/naoqi_driver_node/camera/depth/image_raw' has unsupported encoding '16UC1'
```

nlyubova, 2017-04-06:
@hwp did you try to change the format in Naoqi Driver directly?

JanhviK, 2018-06-18:
@hwp can you tell which rosbag file has to be edited for making changes from 16UC1 to mono16?

frederikscholz, 2018-10-10:
I am currently encountering the same problem, has there been a fix yet, or should I open up a new issue with more details?

hwp, 2018-10-10:
@nlyubova no I did't do that. Although I think the naoqi driver should use the format suggested by REP 118. However, I don't know if changing the format will cause a lot of other problems.

hwp, 2018-10-10:
@JanhviK I did the recording without changing naoqi_driver, and then I modified the rosbag that I recorded.

hwp, 2018-10-10:
@frederikscholz I haven't used depth images for a while. I don't know if it has been fixed.