ros-naoqi/naoqi_driver #91: missing /pepper_robot/camera/depth_registered/points because of empty distortion here

nlyubova, 2017-10-19:
It seems that /pepper_robot/camera/depth_registered/points is empty because commented distortion in naoqi_driver lines 197 and 198 https://github.com/ros-naoqi/naoqi_driver/blob/master/src/converters/camera_info_definitions.hpp#L197 

Should we finally uncomment it to make the pointcloud working?

TuyenNguyenTanViet, 2017-10-25:
its working fine for me :)

mikaelarguedas, 2017-10-25:
My guess for this to be commented out (@Karsten1987 may be able to confirm) is that we didn't have the distortion parameters for the depth camera. The currently commented line is using the distortion values of the RGB camera so uncommenting it without updating the values will likely result in weirdly distorted depth images. So if we have access to the real value it would be worth updating them, otherwise no accurate distortion is better than no image at all :+1:

Regarding the distortion model it may be more future-proof to use the [models defined in sensor_msgs](https://github.com/ros/common_msgs/blob/25613ec79003d8caf0541d7ab36820ed44ebef93/sensor_msgs/include/sensor_msgs/distortion_models.h#L45) rather than a litteral string.

nlyubova, 2017-10-25:
The distortion values are different for each robot. I can provide values for one of them, but it will not be the exact value for every robot. It is better to leave zero distortion.

tomkimsour, 2021-05-26:
I use the commented variables but when i compiled with catkin_make in my ros repository it shows : 

```
src/robobreizh_navigation/naoqi_driver/src/converters/camera_info_definitions.hpp:198:18: error: ambiguous overload for ‘operator=’ (operand types are ‘sensor_msgs::CameraInfo_<std::allocator<void> >::_D_type {aka std::vector<double>}’ and ‘boost::assign_detail::generic_list<double>’)
   cam_info_msg.D = boost::assign::list_of(-0.0688388724945936)(0.0697453843669642)(0.00309518737071049)(
                  ^
In file included from /usr/include/c++/5/vector:69:0,
                 from /usr/include/c++/5/bits/random.h:34,
                 from /usr/include/c++/5/random:49,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/boost/function/detail/prologue.hpp:13,
                 from /usr/include/boost/function.hpp:24,
                 from /opt/ros/kinetic/include/qi/anyfunction.hpp:11,
                 from /opt/ros/kinetic/include/qi/signal.hpp:12,
                 from /opt/ros/kinetic/include/qi/anyobject.hpp:15,
                 from /home/master/Documents/Thomas_ung/robobreizh_pepper_ws/src/robobreizh_navigation/naoqi_driver/include/naoqi_driver/tools.hpp:30,
                 from /home/master/Documents/Thomas_ung/robobreizh_pepper_ws/src/robobreizh_navigation/naoqi_driver/src/converters/converter_base.hpp:24,
                 from /home/master/Documents/Thomas_ung/robobreizh_pepper_ws/src/robobreizh_navigation/naoqi_driver/src/converters/camera.hpp:24,
                 from /home/master/Documents/Thomas_ung/robobreizh_pepper_ws/src/robobreizh_navigation/naoqi_driver/src/converters/camera.cpp:21:
/usr/include/c++/5/bits/vector.tcc:167:5: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(const std::vector<_Tp, _Alloc>&) [with _Tp = double; _Alloc = std::allocator<double>]
     vector<_Tp, _Alloc>::
     ^
In file included from /usr/include/c++/5/vector:64:0,
                 from /usr/include/c++/5/bits/random.h:34,
                 from /usr/include/c++/5/random:49,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/boost/function/detail/prologue.hpp:13,
                 from /usr/include/boost/function.hpp:24,
                 from /opt/ros/kinetic/include/qi/anyfunction.hpp:11,
                 from /opt/ros/kinetic/include/qi/signal.hpp:12,
                 from /opt/ros/kinetic/include/qi/anyobject.hpp:15,
                 from /home/master/Documents/Thomas_ung/robobreizh_pepper_ws/src/robobreizh_navigation/naoqi_driver/include/naoqi_driver/tools.hpp:30,
                 from /home/master/Documents/Thomas_ung/robobreizh_pepper_ws/src/robobreizh_navigation/naoqi_driver/src/converters/converter_base.hpp:24,
                 from /home/master/Documents/Thomas_ung/robobreizh_pepper_ws/src/robobreizh_navigation/naoqi_driver/src/converters/camera.hpp:24,
                 from /home/master/Documents/Thomas_ung/robobreizh_pepper_ws/src/robobreizh_navigation/naoqi_driver/src/converters/camera.cpp:21:
/usr/include/c++/5/bits/stl_vector.h:448:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::vector<_Tp, _Alloc>&&) [with _Tp = double; _Alloc = std::allocator<double>]
       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
       ^
/usr/include/c++/5/bits/stl_vector.h:470:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = double; _Alloc = std::allocator<double>]
       operator=(initializer_list<value_type> __l)
       ^
robobreizh_navigation/naoqi_driver/CMakeFiles/naoqi_driver.dir/build.make:153: recipe for target 'robobreizh_navigation/naoqi_driver/CMakeFiles/naoqi_driver.dir/src/converters/camera.cpp.o' failed
make[2]: *** [robobreizh_navigation/naoqi_driver/CMakeFiles/naoqi_driver.dir/src/converters/camera.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:5140: recipe for target 'robobreizh_navigation/naoqi_driver/CMakeFiles/naoqi_driver.dir/all' failed
make[1]: *** [robobreizh_navigation/naoqi_driver/CMakeFiles/naoqi_driver.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
```

Any idea why ? I only have some basis in c++ so i don't fully understand the error

mbusy, 2021-05-26:
@tomkimsour if I remember correctly you could use [that version](https://github.com/softbankrobotics-research/naoqi_driver/blob/master/src/converters/camera_info_definitions.hpp) of the camera info definition header, it should solve your problem.

Ultimately, we could open a PR to replace the current definition by the one of the fork (getting rid of boost and respecting the c++98 standard), or a more recent one using the standard lib.

tomkimsour, 2021-05-27:
Thanks to your file it now compile properly.

I uninstalled naoqi_driver globaly with ```sudo apt-get install ros-.*-naoqi-driver``` to make sure I was not using it at launch.

This didn't solve my problem but at least now the message looks normal.

UdayRockzz, 2021-11-25:
@ Did you resolve the issue? Even I am getting the same issue when I uncomment those lines(197 and 198)

UdayRockzz, 2021-11-25:
@tomkimsour  Did you resolve the issue? Even I am getting the same issue when I uncomment those lines(197 and 198)

tomkimsour, 2021-11-25:
> @tomkimsour Did you resolve the issue? Even I am getting the same issue when I uncomment those lines(197 and 198)

No i never been able to solve this unfortunatly.