ros-naoqi/naoqi_driver #96: Can't find service: ROS-Driver-Audio

ExpertExcrementExpeditor, 2017-12-06:
Hello, currently trying to get Pepper to run with Kinetic.
Every time I launch `roslaunch pepper_bringup pepper_full.launch nao_ip:=10.215.255.104 roscore_ip:=127.0.0.1`, it crashes after about a minute with the following error:

```
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  	ALProxy::ALProxy
	Can't find service: ROS-Driver-Audio
================================================================================REQUIRED process [pepper_robot/naoqi_driver-1] has died!
process has died [pid 11034, exit code -6, cmd /home/ros-kinetic-box/Development/catkin_ws/devel/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://10.215.255.104:9559 --roscore_ip=127.0.0.1 --network_interface=eth0 --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/ros-kinetic-box/.ros/log/766bfd28-da84-11e7-8a42-0800273e57ed/pepper_robot-naoqi_driver-1.log].
log file: /home/ros-kinetic-box/.ros/log/766bfd28-da84-11e7-8a42-0800273e57ed/pepper_robot-naoqi_driver-1*.log
Initiating shutdown!
================================================================================
```
I've used rosdep to make sure I have all dependencies apart from that, I used the installation tutorial at
http://wiki.ros.org/pepper/Tutorials
switching indigo to kinetic where appropriate.

Full launch log with the log-massages disabled:
```
$ roslaunch pepper_bringup pepper_full.launch nao_ip:=10.215.255.104
sh: 0: getcwd() failed: No such file or directory
... logging to /home/ros-kinetic-box/.ros/log/766bfd28-da84-11e7-8a42-0800273e57ed/roslaunch-roskineticbox-VirtualBox-13782.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://roskineticbox-VirtualBox:41813/

SUMMARY
========

PARAMETERS
 * /pepper_robot/camera/camera_nodelet_manager/num_worker_threads: 4
 * /pepper_robot/camera/depth_rectify_depth/interpolation: 0
 * /pepper_robot/camera/depth_registered_rectify_depth/interpolation: 0
 * /pepper_robot/pose/pose_manager/poses/init/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pepper_robot/pose/pose_manager/poses/init/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/poses/zero/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pepper_robot/pose/pose_manager/poses/zero/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/xap: /opt/ros/kinetic/...
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /pepper_robot/camera/
    bottom_rectify_color (nodelet/nodelet)
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    front_rectify_color (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_front (nodelet/nodelet)
  /pepper_robot/pose/
    pose_controller (naoqi_pose/pose_controller.py)
    pose_manager (naoqi_pose/pose_manager.py)
  /pepper_robot/
    naoqi_driver (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[pepper_robot/naoqi_driver-1]: started with pid [13800]
process[pepper_robot/pose/pose_controller-2]: started with pid [13801]
process[pepper_robot/pose/pose_manager-3]: started with pid [13802]
process[pepper_robot/camera/camera_nodelet_manager-4]: started with pid [13803]
process[pepper_robot/camera/front_rectify_color-5]: started with pid [13804]
process[pepper_robot/camera/bottom_rectify_color-6]: started with pid [13806]
process[pepper_robot/camera/ir_rectify_ir-7]: started with pid [13807]
process[pepper_robot/camera/depth_rectify_depth-8]: started with pid [13823]
Receiving information about robot model
process[pepper_robot/camera/depth_metric_rect-9]: started with pid [13835]
process[pepper_robot/camera/depth_metric-10]: started with pid [13850]
process[pepper_robot/camera/register_depth_front-11]: started with pid [13857]
process[pepper_robot/camera/points_xyzrgb_sw_registered-12]: started with pid [13872]
set prefix successfully to naoqi_driver
[I] 1512568027.051769 13800 qimessaging.session: Session listener created on tcp://0.0.0.0:0
[I] 1512568027.052344 13800 qimessaging.transportserver: TransportServer will listen on: tcp://10.0.2.15:42061
[I] 1512568027.052447 13800 qimessaging.transportserver: TransportServer will listen on: tcp://127.0.0.1:42061
using ip address: 127.0.0.1 @ eth0
process[pepper_robot/camera/depth_registered_sw_metric_rect-13]: started with pid [13880]
process[pepper_robot/camera/depth_registered_rectify_depth-14]: started with pid [13887]
found a catkin prefix /home/ros-kinetic-box/Development/catkin_ws/src/naoqi_driver/share/boot_config.json
load boot config from /home/ros-kinetic-box/Development/catkin_ws/src/naoqi_driver/share/boot_config.json
process[pepper_robot/camera/points_xyzrgb_hw_registered-15]: started with pid [13910]
process[pepper_robot/camera/depth_registered_hw_metric_rect-16]: started with pid [13913]
[ INFO] [1512568027.278369986]: Initializing nodelet with 4 worker threads.
process[pepper_robot/camera/depth_registered_metric-17]: started with pid [13933]
no camera information found for camera_source 3 and res: 1
found a catkin URDF /home/ros-kinetic-box/Development/catkin_ws/src/naoqi_driver/share/urdf/pepper.urdf
[INFO] [1512568028.043074]: Connecting to NaoQi at 10.215.255.104:9559
[I] 1512568028.055423 13801 qimessaging.session: Session listener created on tcp://0.0.0.0:0
[I] 1512568028.055625 13801 qimessaging.transportserver: TransportServer will listen on: tcp://10.0.2.15:42526
[I] 1512568028.055638 13801 qimessaging.transportserver: TransportServer will listen on: tcp://127.0.0.1:42526
[W] 1512568028.055687 14088 qi.path.sdklayout: No Application was created, trying to deduce paths
[WARN] [1512568028.306432]: Collection LLeg not found on your robot.
[WARN] [1512568028.310531]: Collection RLeg not found on your robot.
[INFO] [1512568028.479573]: nao_controller initialized
[INFO] [1512568028.481233]: nao pose_controller running...
[WARN] [1512568030.590660]: stop_walk_srv not available, pose_manager will not stop the walker before executing a trajectory. This is normal if there is no nao_walker running.
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  	ALProxy::ALProxy
	Can't find service: ROS-Driver-Audio
================================================================================REQUIRED process [pepper_robot/naoqi_driver-1] has died!
process has died [pid 13800, exit code -6, cmd /home/ros-kinetic-box/Development/catkin_ws/devel/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://10.215.255.104:9559 --roscore_ip=127.0.0.1 --network_interface=eth0 --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/ros-kinetic-box/.ros/log/766bfd28-da84-11e7-8a42-0800273e57ed/pepper_robot-naoqi_driver-1.log].
log file: /home/ros-kinetic-box/.ros/log/766bfd28-da84-11e7-8a42-0800273e57ed/pepper_robot-naoqi_driver-1*.log
Initiating shutdown!
================================================================================
[pepper_robot/camera/depth_registered_metric-17] killing on exit
[pepper_robot/camera/depth_registered_hw_metric_rect-16] killing on exit
[pepper_robot/camera/points_xyzrgb_hw_registered-15] killing on exit
[pepper_robot/camera/depth_registered_rectify_depth-14] killing on exit
[pepper_robot/camera/depth_registered_sw_metric_rect-13] killing on exit
[pepper_robot/camera/points_xyzrgb_sw_registered-12] killing on exit
[pepper_robot/camera/register_depth_front-11] killing on exit
[pepper_robot/camera/depth_metric-10] killing on exit
[pepper_robot/camera/depth_metric_rect-9] killing on exit
[pepper_robot/camera/depth_rectify_depth-8] killing on exit
[pepper_robot/camera/ir_rectify_ir-7] killing on exit
[pepper_robot/camera/bottom_rectify_color-6] killing on exit
[pepper_robot/camera/front_rectify_color-5] killing on exit
[pepper_robot/camera/camera_nodelet_manager-4] killing on exit
[pepper_robot/pose/pose_manager-3] killing on exit
[pepper_robot/pose/pose_controller-2] killing on exit
[INFO] [1512568059.066730]: Stopping pose_controller
[INFO] [1512568059.066998]: pose_controller stopped
[pepper_robot/naoqi_driver-1] killing on exit
[INFO] [1512568059.698018]: nao pose_controller stopped.
shutting down processing monitor...
... shutting down processing monitor complete
done
```

nlyubova, 2017-12-06:
Hello,
did you install your pepper_bringup with "apt-get install"? If yes, then can you try to compile it from source?

Or you can also try to start naoqi_driver (in a similar way as pepper_bringup)? 
$ roslaunch naoqi_driver naoqi_driver.launch nao_ip:=PEPPER_IP

ExpertExcrementExpeditor, 2017-12-07:
Still getting a similar result:
```
roslaunch naoqi_driver naoqi_driver.launch nao_ip:=10.215.255.104
... logging to /home/ros-kinetic-box/.ros/log/3a495038-db43-11e7-b10e-0800273e57ed/roslaunch-roskineticbox-VirtualBox-2992.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://roskineticbox-VirtualBox:37860/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[naoqi_driver-1]: started with pid [3010]
Receiving information about robot model
set prefix successfully to naoqi_driver
[I] 1512646961.047166 3010 qimessaging.session: Session listener created on tcp://0.0.0.0:0
[I] 1512646961.047494 3010 qimessaging.transportserver: TransportServer will listen on: tcp://10.0.2.15:33820
[I] 1512646961.047572 3010 qimessaging.transportserver: TransportServer will listen on: tcp://127.0.0.1:33820
using ip address: 127.0.0.1 @ eth0
found a catkin prefix /home/ros-kinetic-box/Development/catkin_ws/src/naoqi_driver/share/boot_config.json
load boot config from /home/ros-kinetic-box/Development/catkin_ws/src/naoqi_driver/share/boot_config.json
no camera information found for camera_source 3 and res: 1
found a catkin URDF /home/ros-kinetic-box/Development/catkin_ws/src/naoqi_driver/share/urdf/pepper.urdf
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  	ALProxy::ALProxy
	Can't find service: ROS-Driver-Audio
================================================================================REQUIRED process [naoqi_driver-1] has died!
process has died [pid 3010, exit code -6, cmd /home/ros-kinetic-box/Development/catkin_ws/devel/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://10.215.255.104:9559 --roscore_ip=127.0.0.1 --network_interface=eth0 --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/ros-kinetic-box/.ros/log/3a495038-db43-11e7-b10e-0800273e57ed/naoqi_driver-1.log].
log file: /home/ros-kinetic-box/.ros/log/3a495038-db43-11e7-b10e-0800273e57ed/naoqi_driver-1*.log
Initiating shutdown!
================================================================================
[naoqi_driver-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

nlyubova, 2017-12-07:
did you install it with "apt-get install"? or did you compile them from source?

ExpertExcrementExpeditor, 2017-12-07:
I compiled them from source

nlyubova, 2017-12-07:
can you ping the IP of your robot?

ExpertExcrementExpeditor, 2017-12-07:
Yes.

ExpertExcrementExpeditor, 2017-12-07:
As I said before, i was capable of opening a SSH-connection to the robot and look up the currently running Modules.
Should I try removing the modules I downloaded via apt as well?

nlyubova, 2017-12-07:
You could try to compile naoqi_driver and pepper_bringup from source only.

At first, remove the one installed with apt-get, for example:
$ apt-get remove ros-kinetic-naoqi-driver ros-kinetic-pepper-bringup

then, compile both from source
$ cd catkin_ws/src
$ git clone https://github.com/ros-naoqi/pepper_robot
$ git clone https://github.com/ros-naoqi/naoqi_driver
$ rosdep install -i -y --from-paths ./naoqi_driver
$ cd ..
$ catkin_make

ExpertExcrementExpeditor, 2017-12-07:
Still getting the same results.

I have however noticed this line:
` [W] 1512655031.903106 6150 qi.path.sdklayout: No Application was created, trying to deduce paths
`
might this have something to do with the problems?

nlyubova, 2017-12-07:
The issue most probably not related to Audio, the Audio is just the 1st service to start and when your system crushes it stops on the 1st service that is Audio.

From our previous experience, we had this issue because firewalls were blocking the connection. So it was related to the network problem.

ExpertExcrementExpeditor, 2017-12-12:
Thank you. Apparently the problem was the configuration of the VM.
Using 'NAT' as network configuration led to the aforementioned error.
Using 'Bridged Network' seems to work however. Thank you again for the quick help.

closed 2017-12-12 by ExpertExcrementExpeditor