ros-naoqi/naoqi_driver #101: Getting started with pepper using ros

metalaman, 2018-02-10:
Hello,
I'm new to ROS and wanted help to get [this nod](https://github.com/TsotsosLab/fallen-person-detector)e up and running. The code subscribes to a RBG image, Depth map, Point Cloud and camera info. I have calibrated the rgb camera and saved the yaml file in .ros/camera_info, I could also see pepper's image and depth(b/w) feed from rviz, although I can't see the point cloud, even after uncommenting line 197,198([referred this](https://github.com/ros-naoqi/naoqi_driver/issues/90)).
When I run the fallen person node, nothing happens, the callback taking the rgb image, depth map, camera info and point cloud, never gets called.
How do I make sure that pepper's sensor are sending all those values.
Any help is appreciated?

Pandhariix, 2019-12-06:
Hi @metalaman (nearly 2 years later...) did you manage to retrieve Pepper's point cloud ?

metalaman, 2019-12-11:
I was able to [retrieve Pepper's point](https://github.com/solbach/fallen-person-detector/issues/2) cloud but I don't remember how I got that anymore. Sorry.

Pandhariix, 2019-12-11:
Thanks for the feedback, it's ok. This issue hasn't been updated in a while, I will close it.

closed 2019-12-11 by Pandhariix