ros-naoqi/naoqi_driver #104: naoqi_driver isn't working after reinstallization

illhyhl1111, 2018-02-26:
Hello, we are trying to apply naoqi_driver on **pepper, ros-indigo, Ubuntu 14.04**

Actually, it was working well on our environment at last year.
But after reinstalling ros-indigo and naoqi_driver, it isn't working.

Here is my log.

```
root@bmrr-kimchi:~/catkin_ws# roslaunch naoqi_driver naoqi_driver.launch roscore_ip:=localhost nao_ip:=192.168.1.212
... logging to /home/kimchi/.ros/log/f3b826de-1ac2-11e8-bae1-2c4d54531fb9/roslaunch-bmrr-kimchi-24655.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:45150/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [24667]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f3b826de-1ac2-11e8-bae1-2c4d54531fb9
process[rosout-1]: started with pid [24680]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [24687]
Receiving information about robot model
[W] 24687 qimessaging.remoteobject: Return signature might be incorrect depending on the value, from m to s
[W] 24687 qimessaging.remoteobject: Return signature might be incorrect depending on the value, from m to [[m]]
set prefix successfully to naoqi_driver
[I] 24687 qimessaging.session: Session listener created on tcp://0.0.0.0:0
[I] 24687 qimessaging.transportserver: TransportServer will listen on: tcp://192.168.1.84:35629
[I] 24687 qimessaging.transportserver: TransportServer will listen on: tcp://127.0.0.1:35629
using ip address: localhost @ eth0
found a catkin prefix /home/kimchi/catkin_ws/src/naoqi_driver/share/boot_config.json
load boot config from /home/kimchi/catkin_ws/src/naoqi_driver/share/boot_config.json
no camera information found for camera_source 3 and res: 1
found a catkin URDF /home/kimchi/catkin_ws/src/naoqi_driver/share/urdf/pepper.urdf
Audio Extractor: Start
RightBumperPressed
LeftBumperPressed
BackBumperPressed
ROS-Driver-RightBumperPressed : Start
HandRightBackTouched
HandRightLeftTouched
HandRightRightTouched
HandLeftBackTouched
HandLeftLeftTouched
HandLeftRightTouched
ROS-Driver-HandRightBackTouched : Start
FrontTactilTouched
MiddleTactilTouched
RearTactilTouched
ROS-Driver-FrontTactilTouched : Start
registered subscriber:	teleop
registered subscriber:	moveto
registered subscriber:	speech
[CHECK] keep_looping flag is given
[CHECK] Initialize Done
nodehandle reset 
using master ip: http://localhost:11311
NOT going to re-register the converters
camera/bottom/image_raw is resetting
camera/bottom/image_raw reset
camera/depth/image_raw is resetting
camera/depth/image_raw reset
/diagnostics is resetting
/diagnostics reset
camera/front/image_raw is resetting
camera/front/image_raw reset
imu/base is resetting
imu/base reset
imu/torso is resetting
imu/torso reset
info is resetting
load robot description from file
info reset
camera/ir/image_raw is resetting
camera/ir/image_raw reset
/joint_states is resetting
/joint_states reset
laser is resetting
laser reset
odom is resetting
odom reset
sonar is resetting
sonar reset
resetting subscriber teleop
teleop is resetting
teleop reset
resetting subscriber moveto
moveto is resetting
moveto reset
resetting subscriber speech
speech is resetting
speech reset
resetting service robot config service
robot config service is resetting
resetting service set language service
set language service is resetting
resetting service get language service
get language service is resetting
^C[naoqi_driver-2] killing on exit
[I] 24954 qi.Application: Sending the stop command...
Audio Extractor: Stop
ROS-Driver-RightBumperPressed : Stop
ROS-Driver-HandRightBackTouched : Stop
ROS-Driver-FrontTactilTouched : Stop
Unsubscribe camera handle front_camera_0
Unsubscribe camera handle bottom_camera_0
Unsubscribe camera handle depth_camera_0
Unsubscribe camera handle infrared_camera_0
naoqi driver is shutting down..
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

The process stuck after line 'get language service is resetting'.

Please let me know if something more information needed.
Thanks!

kochigami, 2018-02-26:
Hello, 
I'm sorry that I wrote the code you mentioned.
In my case, it is working in the same environment (Pepper, ros indigo, ubuntu 14.04)...
I'm not sure how to solve it...
I will ask a maintainer to revert it later because I don't have time to look into this for a few weeks...

Could you tell me whether the error message always goes like this?
```
get language service is resetting
^C[naoqi_driver-2] killing on exit
[I] 24954 qi.Application: Sending the stop command...
``` 
Could you tell me which version you have for ``naoqi_driver`` and ``naoqi_bridge_msgs``?
`naoqi_bridge_msgs` version `0.0.8` has been released recently, but I'm not sure why build succeeded in this case.

illhyhl1111, 2018-02-26:
Hello. Thanks for replying!

Well, yes. It always stop on line `get language service is resetting`, and I have to force quit(ctrl+c).
We discovered that it is stuck in a while loop at `Driver::rosLoop()` in `naoqi_driver.cpp`.

the version is
```
root@bmrr-kimchi:~/catkin_ws# rosversion naoqi_driver
0.5.10
root@bmrr-kimchi:~/catkin_ws# rosversion naoqi_bridge_msgs
0.0.8
```
I know that everything is the most recent version.

illhyhl1111, 2018-02-26:
Then, do you recommend to use the past version of naoqi_driver or naoqi_bridge_msgs?

kochigami, 2018-02-26:
Thank you very much for your reply.
I will look into the code tomorrow.

For the time being, could you use the past version of `naoqi_driver`?
I'm sorry for my bad code.

illhyhl1111, 2018-02-26:
Oh god, it is working in version 0.5.9.
Whoa,,, thank you so much.

illhyhl1111, 2018-02-27:
I first installed it by apt-get, and there was some problem with it.
So I downloaded the source and reverted to version 0.5.9 and compiled it, and it worked.

illhyhl1111, 2018-02-27:
By the way, can I ask you just one thing more?
At last year, I used version 0.5.9 with apt-get and the topic which naoqi_driver_node published were something like `/pepper_robot/*` (ex `/pepper_robot/camera/front/image_raw`).
But after building it by source, it publishes like `/naoqi_driver_node/camera/front/image_raw` and so I'm suffering some problems.

Could you please tell me the fastest way to use the published topics with name /pepper_robot/* ?

kochigami, 2018-02-27:
Hello,
about node topic issue, this issue and these pull-requests are related:
https://github.com/ros-naoqi/pepper_robot/issues/35
https://github.com/ros-naoqi/pepper_robot/pull/40
https://github.com/ros-naoqi/naoqi_driver/pull/85

I think you can change the node name by using `pepper_full.launch` in `pepper_robot` or set `ROS_NAMESPACE` in your `.bashrc`.
If we use `pepper_full.launch`, I think we need to get `pepper_robot` from source because the latest release (pepper-robot: 0.1.10) didn't include https://github.com/ros-naoqi/pepper_robot/pull/40.

About the first error (stuck in `get language is resetting`), I couldn't have enough time to look it into today, please wait for a while. I'm sorry.

kochigami, 2018-03-02:
I'm sorry for my one more question.
When you execute `roslaunch naoqi_driver naoqi_driver,launch` with version `0.5.10`, 
is there no result shown by `rosnode list` and `rostopic list`?

If the results are like these (*), I think it is ok.
If not, please keep using `0.5.9'. I can't have access to Pepper for a week. I'm sorry... 
The output on the terminal stops at `get language service is resetting` because of https://github.com/ros-naoqi/naoqi_driver/pull/88.

(*)
`rosnode list`
```
/naoqi_driver
/rosout
```

`rostopic list`
```
/cmd_vel
/diagnostics
/joint_angles
/joint_states
/move_base_simple/goal
/naoqi_driver/audio
/naoqi_driver/bumper
/naoqi_driver/camera/bottom/camera_info
/naoqi_driver/camera/bottom/image_raw
/naoqi_driver/camera/bottom/image_raw/compressed
/naoqi_driver/camera/bottom/image_raw/compressed/parameter_descriptions
/naoqi_driver/camera/bottom/image_raw/compressed/parameter_updates
/naoqi_driver/camera/bottom/image_raw/theora
/naoqi_driver/camera/bottom/image_raw/theora/parameter_descriptions
/naoqi_driver/camera/bottom/image_raw/theora/parameter_updates
/naoqi_driver/camera/depth/camera_info
/naoqi_driver/camera/depth/image_raw
/naoqi_driver/camera/depth/image_raw/compressed
/naoqi_driver/camera/depth/image_raw/compressed/parameter_descriptions
/naoqi_driver/camera/depth/image_raw/compressed/parameter_updates
/naoqi_driver/camera/depth/image_raw/compressedDepth
/naoqi_driver/camera/depth/image_raw/compressedDepth/parameter_descriptions
/naoqi_driver/camera/depth/image_raw/compressedDepth/parameter_updates
/naoqi_driver/camera/depth/image_raw/theora
/naoqi_driver/camera/depth/image_raw/theora/parameter_descriptions
/naoqi_driver/camera/depth/image_raw/theora/parameter_updates
/naoqi_driver/camera/front/camera_info
/naoqi_driver/camera/front/image_raw
/naoqi_driver/camera/front/image_raw/compressed
/naoqi_driver/camera/front/image_raw/compressed/parameter_descriptions
/naoqi_driver/camera/front/image_raw/compressed/parameter_updates
/naoqi_driver/camera/front/image_raw/theora
/naoqi_driver/camera/front/image_raw/theora/parameter_descriptions
/naoqi_driver/camera/front/image_raw/theora/parameter_updates
/naoqi_driver/camera/ir/camera_info
/naoqi_driver/camera/ir/image_raw
/naoqi_driver/camera/ir/image_raw/compressed
/naoqi_driver/camera/ir/image_raw/compressed/parameter_descriptions
/naoqi_driver/camera/ir/image_raw/compressed/parameter_updates
/naoqi_driver/camera/ir/image_raw/theora
/naoqi_driver/camera/ir/image_raw/theora/parameter_descriptions
/naoqi_driver/camera/ir/image_raw/theora/parameter_updates
/naoqi_driver/hand_touch
/naoqi_driver/head_touch
/naoqi_driver/imu/base
/naoqi_driver/imu/torso
/naoqi_driver/info
/naoqi_driver/laser
/naoqi_driver/odom
/naoqi_driver/sonar/back
/naoqi_driver/sonar/front
/rosout
/rosout_agg
/speech
/tf
```

illhyhl1111, 2018-03-02:
Sorry. I left the lab just 20 minutes ago. I will check it on next Monday.

I'm now using the version released in April 2017 and it is fine.  The first 0.5.9 version had some problems in the namespace.

nlyubova, 2018-03-03:
Compiling from source should be fine but you need to do “got pull” in both naoqi_driver and naoqi_bridge_msgs

Regarding the released version (the one that you got with apt-get), you can try installing/updating both naoqi_bridge_msgs and naoqi_driver since they are dependent.

illhyhl1111, 2018-03-05:
`rosnode list` and `rostopic list` is exactly same as you, and I am so sorry to have confused you. 

The problem of our server is, naoqi_driver isn't publishing any topic message.
(`rostopic echo naoqi_driver/camera/front/image_raw` prints nothing)
I thought there is some problem with naoqi_driver, because it prints no log and seems to be stuck in somewhere.
I thought it was fixed after reverting version.. because the log prints out.

I had checked that version 0.5.10 works fine in another server(Ubuntu 16.04), both for apt-get and source compiling.
The first server(Ubuntu 14.04) still isn't working, but I think the fault is in somewhere else.

So sorry once more.;;

kochigami, 2018-03-05:
Don't worry, thank you very much for your report!
In addition, thank you very much for answering my questions again and again!

I'll also check camera topic issue. I'll be back in a week!

nowittyusername, 2018-04-13:
Hi!

Is there any update on this matter? I am getting the same infinite loop on my pepper. It always stops on " get language service is resetting" until I stop the process.

> robot config service is resetting
> resetting service set language service
> set language service is resetting
> resetting service get language service
> get language service is resetting
> ^C[naoqi_driver-2] killing on exit
> [I] 1523606370.456088 26120 qi.Application: Sending the stop command...
> Audio Extractor: Stop

My naoqi_driver and naqi_bridge_msgs versions are 0.5.10 and 0.0.8 respectively.

kochigami, 2018-04-13:
Hi, 

> It always stops on " get language service is resetting" until I stop the process.

I think it is right behavior if `rosnode list` and `rostopic list` show nodes and topics respectively. 
related pull-request:  #88 

Could you please check these commands?

nowittyusername, 2018-04-13:
Thanks for the quick reply!

I tried the commands you suggested and you're right. The driver is actually working in the background.

> ~/ROS/pepper_ws$ rosnode list 
> /naoqi_driver
> /rosout

> ~/ROS/pepper_ws$ rostopic list 
> /cmd_vel
> /diagnostics
> ...
> ...
> /rosout
> /rosout_agg
> /speech
> /tf

However now my question is why doesn't it say anything like `"Driver is ready for use"` or something of the sort. Before the console output comes to a halt at `"get language service is resetting"` there are 5+ outputs of `"resetting X"`, `"X is resetting"` and then finally `"X reset"`. 

As a first time user of your driver I was expecting the same to happen with `get language service`. However it didn't happen no matter how long I waited or restarted ROS, Pepper. Unfortunately, I didn't think of using the commands you provided. 

So shouldn't it give you at least some sort of confirmation of `Hey, I am ready for use`?

kochigami, 2018-04-13:
Thank you very much for your comment.
I am a little bit busy now, but when I have time, I will add some message like "driver is ready for use".
I also appreciate your help.

If you enable "logs" (please see #88 ), you will see endless log message warning.
That is why I send #88, but I did not take care of a first time user at that time. 
Thank you very much for your advice.