ros-naoqi/naoqi_driver #105: naoqi-driver connection drops and freezes every few minutes

StefanWa, 2018-02-26:
After driving around with the Pepper robot for a few minutes, the connection drops (no laser or transform data available anymore) with the following debug output:
```
[DEBUG] [1519656001.875569556]: Connection::drop(2)
[DEBUG] [1519656001.875819948]: Connection::drop(2)
[DEBUG] [1519656001.875915998]: Connection::drop(2)
[DEBUG] [1519656001.908966696]: Socket 15 closed with (ERR|HUP|NVAL) events 24: Broken pipe
[DEBUG] [1519656001.909154913]: TCP socket [15] closed
[DEBUG] [1519656001.909259442]: Connection::drop(0)
[DEBUG] [1519656001.909414087]: Connection to subscriber [callerid=[/move_base] address=[TCPROS connection on port 45687 to [10.10.130.39:45838 on socket 15]]] to topic [/tf] dropped
[DEBUG] [1519656001.909516112]: Connection::drop(2)
[DEBUG] [1519656001.909630790]: Connection::drop(2)
[DEBUG] [1519656001.914136320]: Socket 20 closed with (ERR|HUP|NVAL) events 24: Broken pipe
[DEBUG] [1519656001.914393067]: TCP socket [20] closed
[DEBUG] [1519656001.914448702]: Connection::drop(0)
[DEBUG] [1519656001.914502924]: Connection to subscriber [callerid=[/move_base] address=[TCPROS connection on port 45687 to [10.10.130.39:46554 on socket 20]]] to topic [/naoqi_driver_node/laser] dropped
[DEBUG] [1519656001.914543178]: Connection::drop(2)
[DEBUG] [1519656001.914599753]: Connection::drop(2)
```

If this happens and I want to close down the node I get the following message:
```
^C[naoqi_driver-1] killing on exit
[naoqi_driver-1] escalating to SIGTERM
[naoqi_driver-1] escalating to SIGKILL
Shutdown errors:
 * process[naoqi_driver-1, pid 2327]: required SIGKILL. May still be running
```

From the first output it is clear that this is a network issue, perhaps because the robot is switching access points?
Is there any way to resolve this problem or to be able to recover more gracefully from the issue?

kochigami, 2018-02-27:
Hello,
I'd like to know more information below:
- command (like `roslaunch naoqi_driver naoqi_driver.launch ...`)
- where the command is executed (like `~/catkin_ws/src`)
- whether your `naoqi_driver`/ `naoqi_bridge_msgs` is source or deb (If it is deb, could you tell me the version?)
- ros version
- ubuntu version

Personally, I have never seen this kind of error and I don't know how to solve it.
However, how about connecting your Pepper and your PC directly?
There is some information how to set `PC2NAO` (Local-Link an alternative to DHCP).
http://doc.aldebaran.com/2-1/nao/connectivity.html
If it works, I think the possibility of the network issue increases.

StefanWa, 2018-02-27:
- The command is `roslaunch naoqi-bridge.launch` with the following contents:
```
<launch>
    <arg name="nao_ip"              default="10.10.130.99"/>
    <arg name="nao_port"            default="9559"/>
    <arg name="roscore_ip"          default="10.10.130.39"/>
    <arg name="network_interface"   default="wlp2s0"/>

    <node
        pkg="naoqi_driver"
        type="naoqi_driver_node"
        name="naoqi_driver"
        required="true"
        args="--qi-url=tcp://$(arg nao_ip):$(arg nao_port) --roscore_ip=$(arg roscore_ip) --network_interface=$(arg network_interface)"
    >
    </node>
</launch>
```

- The command is executed in `~/coding/thesis/ros-workspace/launch`.

- `naoqi_driver` is deb and version `0.5.9-0xenial-20180125-135124-0800`
- ROS version is 1.12.12
- Ubuntu version is 16.04 Xenial

Later this week I will try the direct connection using PC2NAO.

Thank you.

nlyubova, 2018-03-03:
The message about “Broken pipe” means that you have lost connection.

How far do you go with your Pepper? 

You could try running naoqi_driver on Pepper.

Pandhariix, 2019-12-06:
Hi @StefanWa, any updates on the issue ?

StefanWa, 2019-12-16:
Hi @Pandhariix, I am not working on this subject anymore.

Pandhariix, 2019-12-16:
Perfect, thanks for the answer :) Closing the issue

closed 2019-12-16 by Pandhariix