ros-naoqi/naoqi_driver #107: Unable to run Naoqi driver C++ version

kamalpy, 2018-04-16:
I am trying to use Pepper with ROS using C++ bridge, when I run the following command:

 `roslaunch pepper_bringup pepper_full.launch  nao_ip:=192.168.8.164 network_interface:=wlp2s0`

I receive the following error and the bridge stops.

```
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  	ALProxy::ALProxy
	Can't find service: ROS-Driver-Audio
```

Please suggest how to resolve this issue.

kamalpy, 2018-04-20:
I am using Naoqi 2.5.3 and ROS is Kinetic but I have also tried with ROS Indigo and getting same error. I need to get the pointcloud that I need to convert to laser scan for gmapping.

nlyubova, 2018-05-03:
@kamalpy the message should be added to the pepper_bringup project

Can you try to launch naoqi_driver with teh same naoi_ip and network_interface? 
roslaunch naoqi_driver naoqi_driver.launch

Pandhariix, 2019-10-10:
Hi @kamalpy. Your issue is similar to the [issue 51](https://github.com/ros-naoqi/pepper_robot/issues/51) of the pepper_robot repo. The solution provided in that issue should solve your problem. Since that issue hasn't been updated in quite some time, I'll close it.

closed 2019-10-10 by Pandhariix