ros-naoqi/naoqi_driver #114: Cannot compile under ROS Melodic and Ubuntu 18.04

gstavrinos, 2019-07-04:
More specifically, the `LIBQI_VERSION` definition cannot be found, but even after manually setting it, a series of errors are encountered.

I have successfully compiled and installed libqi and libqicore,

Thanks in advance, 

George

Pandhariix, 2019-07-05:
Hi Georges, could your logs please ?

gstavrinos, 2019-07-05:
With a fresh clone, the output of `catkin_make` is this:

```
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/recorder/globalrecorder.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/event/audio.cpp:27:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/recorder/globalrecorder.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/event/touch.cpp:27:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/camera.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/camera.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/diagnostics.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/diagnostics.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/info.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/info.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/audio.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/audio.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/helpers/driver_helpers.hpp:22:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/helpers/driver_helpers.cpp:18:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/recorder/globalrecorder.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/recorder/globalrecorder.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/joint_state.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/joint_state.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/imu.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/imu.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/touch.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/touch.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
 #if LIBQI_VERSION>24
                  ^
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/recorder/globalrecorder.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/recorder/recorder.hpp:32,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/naoqi_driver.hpp:45,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/naoqi_driver.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:18: error: operator '>' has no left operand
[... 1796 lines cut ...]
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:30,
[... cut ...] are, _Alloc>::const_iterator = std::_Rb_tree_const_iterator<std::pair<const std::__cxx11::basic_string<char>, boost::shared_ptr<urdf::JointMimic> > >; std::map<_Key, _Tp, _Compare, _Alloc>::value_type = std::pair<const std::__cxx11::basic_string<char>, boost::shared_ptr<urdf::JointMimic> >]
       insert(const_iterator __position, const value_type& __x)
       ^~~~~~
/usr/include/c++/7/bits/stl_map.h:852:7: note:   candidate expects 2 arguments, 1 provided
/usr/include/c++/7/bits/stl_map.h:862:7

Pandhariix, 2019-07-06:
Alright. __LIBQI_VERSION__ is defined with __naoqi_libqi_VERSION_MAJOR__ and __naoqi_libqi_VERSION_MINOR__, defined by the __naoqi_libqi__ ros package. The package can be installed for indigo and kinetic, but the melodic version is not yet available...

Only compiling and installing libqi and libqicore wouldn't solve the problem, the best solution would be to release them as ros packages for melodic ([image] and [image]). But in the meantime, I don't see why manually setting __LIBQI_VERSION__ doesn't work, how did you do it ?

gstavrinos, 2019-07-08:
I tried just overriding the ifdef blocks by allowing either of the two calls to be executed, but neither of the two actually worked, each one resulting in a different set of errors. I can provide you with my logs if you want to.

Thanks.

Pandhariix, 2019-07-12:
I see... Sure, logs are always welcome!

gstavrinos, 2019-07-22:
I am sorry for the delay. I had to catch up with other issues! Here is my slightly long output:
```

In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/recorder/globalrecorder.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/recorder/globalrecorder.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
<command-line>:0:0: note: this is the location of the previous definition
[ 88%] Built target moveit_ros_control_interface_trajectory_plugin
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/recorder/globalrecorder.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/event/audio.cpp:27:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/touch.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/touch.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/helpers/driver_helpers.hpp:22:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/helpers/driver_helpers.cpp:18:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/imu.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/imu.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/camera.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/camera.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/audio.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/audio.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
<command-line>:0:0: note: this is the location of the previous definition
[ 88%] Built target moveit_ros_control_interface_plugin
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/recorder/globalrecorder.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/event/touch.cpp:27:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/info.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/info.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24:0,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/diagnostics.hpp:24,
                 from /home/gstavrinos/catkin_ws/src/naoqi_driver/src/converters/diagnostics.cpp:21:
/home/gstavrinos/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:65:0: warning: "LIBQI_VERSION" redefined
 #define LIBQI_VERSION 26
 
[... 1816 lines cut ...]
 inline bool operator!=(detail::function::useless_clear_type*,
             ^~~~~~~~
/usr/include/boost/function/function_base.hpp:721:13: note:   no known conversion for argument 1 from ‘std::map<std::__cxx11::basic_string<char>, boost::shared_ptr<urdf::Joint> >::iterator {aka std::_Rb_tree_iterator<std::pair<const std::__cxx11::basic_string<char>, boost::shared_ptr<urdf::Joint> > >}’ to ‘boost::detail::function::useless_clear_type*’
[... cut ...] :__cxx11::basic_string<char>, boost::shared_ptr<urdf::JointMimic> >]
       insert(value_type&& __x)
       ^~~~~~
/usr/include/c++/7/bits/stl_map.h:802:7: note:   no known conversion for argument 1 from ‘std::pair<std::__cxx11::basic_string<char>, std::shared_ptr<urdf::JointMimic> >’ to ‘std::map<std::__cxx11::basic_string<char>, boost::shared_ptr<urdf::JointMimic> >::value_type&& {aka std::pair<const std::__cxx11::basic_string<char>, boost::shared_ptr<urdf::JointMimic> >&&}’
/usr/include/c++/7

mbusy, 2020-02-28:
(Very late answer) naoqi_libqi, naoqi_liqicore (along with with naoqi_driver, naoqi_bridge_msgs and the robots meshes) have been released for ROS melodic. You should be able to compile naoqi_driver for melodic, or even install it using apt-get

closed 2020-02-28 by mbusy

gstavrinos, 2020-03-11:
Hey @mbusy , thanks for reaching out. Do we have to run a specific software version on our Pepper?

mbusy, 2020-03-11:
Hi @gstavrinos, do you mean the NAOqi version you should use? NAOqi 2.5.x should be fine (but it should also be compatible with NAOqi 2.4).

gstavrinos, 2020-03-11:
That's what I meant, because I think there were problems with some NAOqi versions. Thanks again!

mbusy, 2020-03-11:
You're welcome!