ros-naoqi/naoqi_driver #115: How to improve the depth image frame rate?

kuku6118, 2019-07-15:
Hi,
We use "rostopic hz /naoqi driver/camera/depth/image_raw" to get a depth image with an average rate of only 2. Is there a way to improve this problem?

We have tried the following adjustments are not feasible:
1. use rostopic hz /naoqi driver/camera/depth/image_raw/compressed,
2. disable front_camera, bottom_camera, and stereo_camera

By the way, where is the code for the wifi receiver? We can also test the rate there.

nlyubova, 2019-07-15:
Hi,
Compressed image should help. I've used it before and it allowed to increase the rate.
Try to decrease resolution and disable IR image transfer. 

Still, the main problem is WiFi.
With Ethernet cable, it should be better

kuku6118, 2019-07-16:
@nlyubova We link the ethernet cable to the pepper head and PC side. How to use Ethernet to connect with Pepper?

kuku6118, 2019-07-23:
@nlyubova We have changed the WiFi to the Ethernet cable with significant improvement.
Thanks for helping

closed 2019-07-23 by kuku6118

ZiyunGe1999, 2019-11-15:
> @nlyubova We link the ethernet cable to the pepper head and PC side. How to use Ethernet to connect with Pepper?

Hi, I'm trying to increase the image frame rate too (through Wifi not ethernet cable). Would you please tell me how high your final frame rate is after you had tried so many ways?