ros-naoqi/naoqi_driver #117: Did not get IMU data

kuku6118, 2019-07-23:
Hi
We learned that the pepper has two IMU sensors located in the chest and chassis.
But after the individual rostopic echo, the following message
1. rostopic echo /naoqi_driver/imu/base
[image]
2. rostopic echo/naoqi_driver/imu/torso
[image]

There are two problems:
1. rostopic echo/naoqi_driver/imu/torso clearly did not get IMU data.
    A log similar to the following occurs
   [W] 1563862311.485091 32079 qitype.genericvalue: Conversion from v(void) to f(float) failed
Conversion from v(void) to f(float) failed=> set to -1
2. Did not provide all the covariance information. Is there a way to solve this?

Pandhariix, 2019-07-26:
Hi,
Regarding your first point, on the Pepper robot the only accessible IMU is located in the base of the robot (specified [here](http://doc.aldebaran.com/2-5/family/pepper_technical/inertial_pep.html#location)). On the NAO robot, the only accessible IMU is located in the torso of the robot (as mentioned [here](http://doc.aldebaran.com/2-5/family/robots/inertial_robot.html)).

The values for Pepper's IMU are retrieved in [__imu.cpp__](https://github.com/ros-naoqi/naoqi_driver/blob/master/src/converters/imu.cpp), using the memory keys specified [there](http://doc.aldebaran.com/2-5/family/pepper_technical/pepper_dcm/actuator_sensor_names.html#ju-inertial-base). 

I totally get your confusion though, naoqi_driver will publish torso IMU values even if you're connected to Pepper. This [snippet](https://github.com/ros-naoqi/naoqi_driver/blob/fbda38eb35e678cc0d8f9b7a17108012e870d6d7/src/naoqi_driver.cpp#L659) is responsible for that, we should have the following snippet instead:
```cpp
if(robot_ == robot::NAO)
{
  /** IMU TORSO **/
  if ( imu_torso_enabled )
  {
```

Regarding your second issue, in its current situation, naoqi_driver will publish the estimation values depicted in your logs (see this [snippet](https://github.com/ros-naoqi/naoqi_driver/blob/fbda38eb35e678cc0d8f9b7a17108012e870d6d7/src/converters/imu.cpp#L123)), following the directives of [sensor_msgs/Imu](http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/Imu.html). Unfortunately I don't think that there is an easy way to get those covariances...

kuku6118, 2019-07-30:
@Pandhariix Thank you for all your assistance. It seems that we can only give up the IMU of the torso first.

closed 2019-07-30 by kuku6118