ros-naoqi/naoqi_driver #119: how to install naoqi_libqicore?

3038922, 2019-09-30:
my system is ubuntu1804 
ros --version is melodic .
if i input `rosdep install -i -y --from-paths ./naoqi_driver`
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
naoqi_driver: Cannot locate rosdep definition for [naoqi_libqicore]
```

Pandhariix, 2019-10-02:
Hi,

The naoqi_libqicore package has not yet been released for melodic, you won't be able to get it through rosdep or apt-get.

3038922, 2019-10-11:
https://github.com/aldebaran/libqicore ->naoqi_libqicore
https://github.com/aldebaran/libqi -> naoqi_libqi
I am a novice, can you tell me how to compile libqicore? How to compile is not written in the document

Pandhariix, 2019-10-13:
To compile those packages for naoqi_driver, you would need to modify them so that they match the format of ROS packages (named naoqi_libqi and naoqi_libqicore instead of libqi and libqicore), put them in the catkin workspace of your naoqi_driver repo and compile everything with catkin.

But this is not an easy task, extra modifications should be brought to libqi and libqicore in order to compile them for ubuntu 18 (mainly because of boost compatibility issues). I've opened an [issue](https://github.com/ros-naoqi/libqi-release/issues/6) regarding the release of naoqi_libqi and naoqi_libqicore for melodic. With those 2 packages released, compiling naoqi_driver for melodic should be way easier.

mbusy, 2020-02-28:
This is a bit of a necropost, but naoqi_libqicore (along with naoqi_driver, naoqi_bridge_msgs and the robots meshes) has been released for ros melodic

closed 2020-02-28 by mbusy