ros-naoqi/naoqi_driver #120: Error : Do not calculate NAO Footprint, no transform possible

Meli888, 2019-10-04:
Hello, I am working on Ubuntu 16.04 with Kinetic version of Ros. I have a Nao V6. 
When I run a roscore and roslaunch naoqi_driver naoqi_driver.launch (with corresponding nao_ip, roscore_ip et network_interface), I have this error : "Do not calculate NAO Footprint, no transform possible". I cannot move the joints of the Nao with this error.
Thanks

Pandhariix, 2019-10-07:
Hi, indeed that has to be fixed. But it shouldn't prevent you from moving NAO's joints, using the /joint_angles topic (if the joints are stiffed). For instance, that message should move the robot's head:
```shell
rostopic pub /joint_angles naoqi_bridge_msgs/JointAnglesWithSpeed "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
joint_names: ['HeadPitch']
joint_angles: [0.2]
speed: 0.2
relative: 0"
```
Can you try to publish that exact message and get back to me ?
Thanks

Meli888, 2019-10-07:
Hi, thank you for your answer !
You're right, I can move the robot with your command.
The error is always there.

Pandhariix, 2019-10-07:
Perfect! 

Yep, this error has actually been there for quite some time, it is triggered because of a failing [canTransform](https://github.com/ros-naoqi/naoqi_driver/blob/fbda38eb35e678cc0d8f9b7a17108012e870d6d7/src/converters/nao_footprint.hpp#L42). It should be investigated, my guess is that it could be linked to the cache_time of the tf2 buffer. 

In any case, if you can move the joints of the robot, you shouldn't be impacted!