ros-naoqi/naoqi_driver #133: Cannot compile under ROS Melodic in MacOS

iPsych, 2020-02-28:
In MacOS with ROS Melodic successfully installed, below error happens.
I tried to tap ros/melodic but it shows "remote: Repository not found." Error.

```
Error: No available formula with the name "ros/melodic/naoqi_bridge_msgs" 
Please tap it and then try again: brew tap ros/melodic
Error: No available formula with the name "ros/melodic/naoqi_libqicore" 
Please tap it and then try again: brew tap ros/melodic
Error: No available formula with the name "ros/melodic/naoqi_libqi" 
Please tap it and then try again: brew tap ros/melodic
executing command [brew install ros/melodic/naoqi_bridge_msgs]
==> Tapping ros/melodic
Cloning into '/usr/local/Homebrew/Library/Taps/ros/homebrew-melodic'...
remote: Repository not found.
fatal: repository 'https://github.com/ros/homebrew-melodic/' not found
Error: Failure while executing; `git clone https://github.com/ros/homebrew-melodic /usr/local/Homebrew/Library/Taps/ros/homebrew-melodic` exited with 128.
ERROR: the following rosdeps failed to install
  homebrew: command [brew install ros/melodic/naoqi_bridge_msgs] failed
```

mikaelarguedas, 2020-02-28:
AFAIK there is no tap or bottles for ROS.
How did you install the rest of ROS Melodic ?

most likely you need to build from source all the dependencies of the package you are trying to build.

One convenient way to get a list of repos to download based on what you already have installed is using rosinstall_generator:

```
pip install rosinstall_generator vcstool
source /path/to/ros/melodic/setup.bash
mkdir -p /path/to/ws/src && cd /path/to/ws/
rosinstall_generator naoqi_driver --deps --exclude RPP --tar --rosdistro melodic > naoqi_driver_deps.rosinstall
vcs import src < naoqi_driver_deps.rosinstall
```

then you can install dependencies (using rosdep) and build like you were likely doing when encountering the error

iPsych, 2020-02-29:
@mikaelarguedas Thanks.
For the remaining ROS install, I followed the instruction of below URL and succeed. https://gist.github.com/ahoarau/dd521343494d9ec3ffb25da3f61f346d

I followed your instructions, but face below error.
```
make[1]: *** [naoqi_libqi/CMakeFiles/qi.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 9 jobserver tokens available; should be 8!
Invoking "make -j8 -l8" failed
```
Which happened in below
```
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__string:57:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2494:5: error: static_assert failed due to requirement
      '__is_forward_iterator<boost::iterators::transform_iterator<unsigned long (*)(const boost::basic_string_ref<char, std::__1::char_traits<char> > &), std::__1::__wrap_iter<const
      boost::basic_string_ref<char, std::__1::char_traits<char> > *>, boost::iterators::use_default, boost::iterators::use_default> >::value' "std::max_element requires a ForwardIterator"
    static_assert(__is_forward_iterator<_ForwardIterator>::value,
    ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2512:19: note: in instantiation of function template specialization
      'std::__1::max_element<boost::iterators::transform_iterator<unsigned long (*)(const boost::basic_string_ref<char, std::__1::char_traits<char> > &), std::__1::__wrap_iter<const
      boost::basic_string_ref<char, std::__1::char_traits<char> > *>, boost::iterators::use_default, boost::iterators::use_default>, std::__1::__less<unsigned long, unsigned long> >' requested
      here
    return _VSTD::max_element(__first, __last,
                  ^
/usr/local/include/boost/range/algorithm/max_element.hpp:44:17: note: in instantiation of function template specialization 'std::__1::max_element<boost::iterators::transform_iterator<unsigned
      long (*)(const boost::basic_string_ref<char, std::__1::char_traits<char> > &), std::__1::__wrap_iter<const boost::basic_string_ref<char, std::__1::char_traits<char> > *>,
      boost::iterators::use_default, boost::iterators::use_default> >' requested here
    return std::max_element(boost::begin(rng), boost::end(rng));
                ^
/Users/eri/catkin_ws/src/naoqi_libqi/src/print.cpp:355:29: note: in instantiation of function template specialization 'boost::range::max_element<boost::range_detail::transformed_range<unsigned
      long (*)(const boost::basic_string_ref<char, std::__1::char_traits<char> > &), const std::__1::vector<boost::basic_string_ref<char, std::__1::char_traits<char> >,
      std::__1::allocator<boost::basic_string_ref<char, std::__1::char_traits<char> > > > > >' requested here
    const auto maxLabelIt = max_element(labels | transformed(boost::size<string_ref>));

```
```
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__string:57:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2494:5: error: static_assert failed due to requirement
      '__is_forward_iterator<boost::iterators::transform_iterator<unsigned long (*)(const std::__1::basic_string<char> &), std::__1::__wrap_iter<const std::__1::basic_string<char> *>,
      boost::iterators::use_default, boost::iterators::use_default> >::value' "std::max_element requires a ForwardIterator"
    static_assert(__is_forward_iterator<_ForwardIterator>::value,
    ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2512:19: note: in instantiation of function template specialization
      'std::__1::max_element<boost::iterators::transform_iterator<unsigned long (*)(const std::__1::basic_string<char> &), std::__1::__wrap_iter<const std::__1::basic_string<char> *>,
      boost::iterators::use_default, boost::iterators::use_default>, std::__1::__less<unsigned long, unsigned long> >' requested here
    return _VSTD::max_element(__first, __last,
                  ^
/usr/local/include/boost/range/algorithm/max_element.hpp:44:17: note: in instantiation of function template specialization 'std::__1::max_element<boost::iterators::transform_iterator<unsigned
      long (*)(const std::__1::basic_string<char> &), std::__1::__wrap_iter<const std::__1::basic_string<char> *>, boost::iterators::use_default, boost::iterators::use_default> >' requested here
    return std::max_element(boost::begin(rng), boost::end(rng));
                ^
/Users/eri/catkin_ws/src/naoqi_libqi/src/print.cpp:320:30: note: in instantiation of function template specialization 'boost::range::max_element<boost::range_detail::transformed_range<unsigned
      long (*)(const std::__1::basic_string<char> &), const std::__1::vector<std::__1::basic_string<char>, std::__1::allocator<std::__1::basic_string<char> > > > >' requested here
      const auto maxElemIt = max_element(elementNames | transformed(boost::size<std::string>));
```

mikaelarguedas, 2020-03-02:
Ok so you made progress and you are now failing the build on naoqi_libqi.

Unfortunately maintainers here are not maintainers of libqi and the packages here have been built only on Ubuntu Bionic with no attempt to support macos.

You may have more luck posting your question directly at: https://github.com/aldebaran/libqi

---

The error you are facing doesn't seem to be ROS related but rather boost and xcode related

Some leads to explore:
- what version of boost are you using ? your installation scripts says boost 1.69 but the code failing has been removed in boost 1.68 https://github.com/boostorg/range/commit/adcb071dc6b06e69aa4ddcc080d23695a5da378f
- what version of XCode and c++ standard are you using ?

iPsych, 2020-03-08:
@mikaelarguedas Thanks.
I am repeatedly face to /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2494:5: error: static_assert failed due to requirement
Which boost version was used in Linux compilation?

mikaelarguedas, 2020-03-08:
The debian packages are built using boost 1.65.1 (version provided by Ubuntu 18.04)

iPsych, 2020-03-10:
@mikaelarguedas 
I found it seems a clang bug 
`error: static_assert failed due to requirement '__is_forward_iterator<boost::adj_list<unsigned long>::edge_iterator>::value' "std::max_element requires a ForwardIterator" :info:build     static_assert(__is_forward_iterator<_ForwardIterator>::value, 
`
and discussed as below.

[https://git.skewed.de/count0/graph-tool/-/merge_requests/19]

[(https://www.mail-archive.com/llvm-bugs@lists.llvm.org/msg30845.html)]