ros-naoqi/naoqi_driver #134: ros in naoV6

zxvp, 2020-08-20:
Hello, I want to use NAOV6 in ROS, but the tutorial on the official website is not suitable. What should I do. Thank you

mbusy, 2020-08-24:
Hi, which ROS distro are you using? You should install the following packages :
- ros-yourdistro-naoqi-libqi
- ros-yourdistro-naoqi-libqicore
- ros-yourdistro-nao-meshes
- ros-yourdistro-naoqi-bridge-msgs
- ros-yourdistro-naoqi-driver

Those are the "basic packages", allowing you to interact with the NAO (V5 or V6) robot. They have been released for Melodic and for lesser versions (not entirely for Lunar though).

You can then launch the naoqi_driver package to work with your NAO in ROS:
```shell
# You can get your network interface by typing the ifconfig command in your console 
roslaunch naoqi_driver naoqi_driver.launch nao_ip:=your_robot_ip network_interface:=your_interface
```

mbusy, 2020-11-13:
@zxvp Did you manage to connect your V6 to ROS ?

mbusy, 2022-09-15:
Closing the issue for lack of activity

closed 2022-09-15 by mbusy