ros-naoqi/naoqi_driver #137: The sensor data updates really slow

yhan33, 2020-11-13:
Hi, I successfully connected my Nao H25 V5 using naoqi_driver. However, as I rostopic echo /joint states, the data updated very slow for some reason. Can I get some help here?

Thanks

mbusy, 2020-11-13:
Hi @yhan33, could you specify what you mean by "really slow" in terms of fps ? Also, I assume that naoqi_driver is running on your computer, how are your computer and your robot connected to the network (WiFi or Ethernet) ?

yhan33, 2020-11-13:
Thanks for the reply.

I am using NAO H25 V5. The robot connects to the PC via Ethernet. 

I have a couple problems and one of them are mentioned in the previous issues.

1) There is always an error when I launch the node, as shown in the following picture
[image]

2)After roslaunch, I did "rostopic echo/joint_states". It seems the hand joint angle measurements are not included since the hand joint angles are "nan", as shown in the following picture
[image]

3) "Really slow" means the scanning frequency of the topic joint_state is very low. I looked into the naoqi_driver.cpp file and it shows that the joint_states frequency is 20 Hz (as shown in the follow picture). I think I only get less than 5 Hz.
[image]

Thanks again for your help.

mbusy, 2020-11-16:
So if I understand correctly, the robot is directly connected to the computer through Ethernet then ?

Regarding the first two issue:
1.  That's indeed a known bug (check [here](https://github.com/ros-naoqi/naoqi_driver/issues/120#issuecomment-538924106)), but it shouldn't prevent you from doing what you're trying to achieve
2. That is actually normal, as the "hand joints" are different from the other revolute joints of the robot. The hand joints position will be a value between 0 and 1 (corresponding to a percentage of hand opening/closing). The velocity of that "joint" and its effort are not computed, hence resulting in a nan in the joint_states message

Regarding your third point, that's strange, it should be faster. First, make sure that you also have 20Hz in the `share/boot_config.json` file. Then, are you running things in parallel on your computer / on the robot ?

yhan33, 2021-02-09:
> So if I understand correctly, the robot is directly connected to the computer through Ethernet then ?
>
> Regarding the first two issue:
>
> 1. That's indeed a known bug (check [...]

Thanks for your reply. 
I have changed the joint_state rate in boot_config.json  file (see attached the snapshot)
[image]

It seems the maximum frequency is always 13.3Hz (see the snapshot)
[image]

> Then, are you running things in parallel on your computer / on the robot ?
No, I am only running the naoqi_driver to read the sensor data from the robot

Can you please point out if I can change anywhere else to get around 50 Hz sensor frequency?

Thanks

mbusy, 2021-02-11:
Unfortunately, you won't be able to get a better frequency than what you already have... Less than 5Hz seemed really low, but that frequency seems correct

yhan33, 2021-02-11:
Anyway. Thanks for your kind help!

mbusy, 2021-02-11:
@yhan33 You're welcome!

closed 2021-02-11 by mbusy