ros-naoqi/naoqi_driver #138: Will there be Noetic support?

holken, 2021-02-12:
Hello, I'm currently on Ubuntu focal which isn't supported by previous versions of ROS. I also noticed that naoqi driver isn't supporting Noetic either. So I'm stuck between a rock and a hard place.

Is there any plan to support Noetic or is the only option to downgrade Ubuntu and ROS?

mbusy, 2021-02-12:
@holken Yes, we have plans to release the driver for Noetic. I wouldn't be able to give you a release date though, the first packages to be released for Noetic will be naoqi_libqi and naoqi_libqicore, followed by naoqi_driver and naoqi_bridge_msgs.

For now, the most reasonable option would be downgrading Ubuntu and ROS. You mentioned running NAO virtually: another solution __(not yet tested on our side, and if you only want to work with a virtual robot)__ would be to use [qibullet](https://github.com/softbankrobotics-research/qibullet) and its ROS stack, it should be compatible with focal and Noetic

tomkimsour, 2022-03-01:
Now that libqi and libqicore are out. Do you know if there is a time frame where noetic support for the driver would be release ?

mbusy, 2022-09-15:
To close the issue,
* naoqi_libqi
* naoqi_libqicore
* naoqi_bridge_msgs
* nao_meshes
* pepper_meshes

Are currently being released for Noetic and will soon be available throught apt-get (corresponding [sync topic](https://discourse.ros.org/t/preparing-for-noetic-sync-2022-09-14/27292)). 
The naoqi_driver package is being released too, but will be included in the next batch. Anyways, it should soon be available too, check the ROS [Discourse](https://discourse.ros.org/c/release/noetic/66) release schedule for Noetic

closed 2022-09-15 by mbusy

mbusy, 2022-10-03:
Hey @sasilva1998, yes it should be enough. To make things clear, the whole stack isn't going to be maintained anymore, only the driver, its dependencies and the mesh packages.

So naoqi_driver, naoqi_libqi, naoqi_libqicore, naoqi_bridge_msgs and the pepper_meshes and nao_meshes packages are maintained, not the rest. It doesn't mean that you cannot try to compile it under noetic to interact with a robot, but it won't be maintained. In your case, can I ask, what robot are you using? (and which naoqi version)

sasilva1998, 2022-10-03:
@mbusy I am using a Pepper Robot. I was just testing, and noticed that I do get the positions of the robot joints in rviz even though `naoqi_poses.py` crashes because of the python version difference of the sdk, not sure what `naoqi_poses.py` does though.

I will be trying to make `naoqi_dcm_driver` compile to be able to use moveit.

sasilva1998, 2022-10-05:
Hi @mbusy, in the end I could make it work, and was able to use moveit and teleop with the real pepper robot.

mbusy, 2022-10-06:
Great! If you're motivated to open a PR on the project it's even greater, I can't promise that it will be reviewed soon though 😅

adrianrodriguez2016, 2024-01-21:
Hi @sasilva1998  

1. how did you end up fixing your issues with the python version? Did you install python2.7 for this?

2. What did you do to be able to compile the naoqi_dcm_driver for noetic?

victorpaleologue, 2024-01-22:
Hello @adrianrodriguez2016, you are not meant to require Python 2.7 and naoqi_dcm_driver to get it work for Noetic.
Please review the README (which may have been updated since then), and file an issue with your problem to get help.