ros-naoqi/naoqi_driver #139: Slow image topics

alex-kozinov, 2021-04-27:
Hi!
I recently managed to install ROS directly on NAO with your [help](https://github.com/ros-naoqi/naoqi_driver/issues/122#issuecomment-592607224). But now i'm trying to run naoqi on robot. And the case is that topics `/naoqi_driver/camera/bottom/image_raw` and `naoqi_driver/camera/front/image_raw` have low frame rate:
* front images
[image]
* bottom images
[image]

Do you know how to increase this rate up to 30 frames per second?

P.s. i've run naoqi with command
```
roslaunch naoqi_driver naoqi_driver.launch nao_ip:=$ROS_IP network_interface:=wlan0
```

And i tried to increase "fps" parameter at boot_config.json for both cameras:
[image]

mbusy, 2021-05-26:
Hi, I assume that you are using the `sbre_robot_ros_gentoo_prefix` project mentioned in #122?

alex-kozinov, 2021-05-31:
Yes, i am :)