ros-naoqi/naoqi_driver #141: Pepper base_link is located in his torso

tomkimsour, 2021-05-18:
Hello,

I'm running a mapping algorithm on rviz but i have problem with pepper transformation.

Currently running on ubuntu 16.04 with ros kinetic.

I have a problem, when visualizing in rviz, I get many errors of this type in tf :

```no transform from [pepper robot/Bumper frame] to frame [odom]```

Also I can see the base_link is not located at pepper's foot.

Does anyone have a clear explaination of what is happening ?

To me it looks like it's coming from the driver who doesn't handle odometry correctly.

mbusy, 2021-05-26:
Hi, regarding the first part, it could be related to your rviz config. You could try and replace your config (make a backup) in `~/.rviz` by [this file](https://github.com/ros-naoqi/naoqi_driver/blob/master/share/pepper.rviz) (you'll have to rename it with the name of your previous config).

More generally, the transform between the base link and the odom can be specified in the launchfile (see [this example](https://github.com/ros-naoqi/pepper_moveit_config/blob/3efef9c446b8426425b45829bde028c1e514bd7e/launch/demo.launch#L18))

For the second part, this is due to the fact that the naoqi framework places the "base link" of Pepper in its torso (as you can see [here](http://doc.aldebaran.com/2-5/family/pepper_technical/links_pep.html)). If you are seeking a modified Pepper URDF (with a base_link located in the base footprint of the robot), you could use softbankrobotics-research's [fork](https://github.com/softbankrobotics-research/naoqi_driver) of naoqi_driver. 

You should although be able to properly display the Pepper robot in RViz with this repo.

tomkimsour, 2021-05-27:
Thanks for your answer. This one really helped me.

> For the second part, this is due to the fact that the naoqi framework places the "base link" of Pepper in its torso (as you can see here). If you are seeking a modified Pepper URDF (with a base_link [...]

Also for the error I don't properly remember how i manage to have it but i solved it by tweaking some parameters which i can't recall either.

closed 2021-05-27 by tomkimsour