ros-naoqi/naoqi_driver #142: catkin error on jetson TX2

xiejialong, 2021-06-10:
When I compile the package naoqi_driver on Jetson TX2 with kinetic, a catkin error is reported as follows:

[100%] Built target naoqi_driver_module
/opt/ros/kinetic/lib/libqi.so: undefined reference to `__atomic_compare_exchange_16'
/opt/ros/kinetic/lib/libqi.so: undefined reference to `__atomic_store_16'
/opt/ros/kinetic/lib/libqi.so: undefined reference to `__atomic_load_16'
collect2: error: ld returned 1 exit status
naoqi_driver/CMakeFiles/naoqi_driver_node.dir/build.make:182: recipe for target '/home/nvidia/catkin_ws/devel/lib/naoqi_driver/naoqi_driver_node' failed
make[2]: *** [/home/nvidia/catkin_ws/devel/lib/naoqi_driver/naoqi_driver_node] Error 1
CMakeFiles/Makefile2:477: recipe for target 'naoqi_driver/CMakeFiles/naoqi_driver_node.dir/all' failed
make[1]: *** [naoqi_driver/CMakeFiles/naoqi_driver_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

I guess whether because of the arm64 for TX2? . In contrast,  the package can be compiled in my common computer

mbusy, 2021-07-05:
Hi, indeed the arm architecture makes the compilation of the naoqi ros stack quite interesting on a Jetson TX2 😄 I understand from your message that you managed to compile the naoqi-libqi and naoqi-libqicore packages for the Jetson ?

xiejialong, 2021-07-05:
yes, several io files cannot be compiled on Jetson TX2. sometimes, some other software also exists this problem.

mbusy, 2021-07-05:
I think that [this issue](https://github.com/aldebaran/libqi-python/issues/9#issuecomment-707097741) could contain the solution to your problem (the libqi patch mentioned in the comment in particular). Adding the following line:
```shell
find_package(GccAtomic)
```

To the `CMakeLists.txt` of naoqi-libqi and recompiling the package might do the trick. Is the naoqi-libqi package in your current catkin workspace?

mbusy, 2021-07-05:
Also, you will need to create a `FindGccAtomic.cmake` file, under a `cmake` folder at the root of the naoqi-libqi package

xiejialong, 2021-07-05:
Thank you very much for your suggestion. Once I go back to my Lab, I will certainly try it and tell you the result.  😁

mbusy, 2022-09-15:
@xiejialong any updates? If not I'll close the issue for now

xiejialong, 2022-09-15:
I haven’t any updates. Thanks

mbusy, 2022-09-15:
Alright, I'll close the issue

closed 2022-09-15 by mbusy