ros-naoqi/naoqi_driver #143: ros2 porting

giodegas, 2021-07-03:
Is there any plan to port this project to ROS2?

mbusy, 2021-07-05:
Hi, yes absolutely, although we don't really have a specific time-frame for its release (we still have a few backward compatibility problems to fix). 

If you want to use what has been developed so far (depending on your setup the stability of the solution might vary), check the ros2-intergration branch of [this fork](https://github.com/mbusy/naoqi_driver/tree/ros2_integration). Ultimately, we want to merge those features into this repo, and release the package

mbusy, 2022-09-15:
To close this issue, the essential part of the stack has been ported to ROS2, and some releases are pending. The following packages have been ported:
* naoqi_libqi ([libqi](https://github.com/ros-naoqi/libqi/tree/ros2) repo, ros2 branch)
* naoqi_libqicore ([libqicore](https://github.com/ros-naoqi/libqicore/tree/ros2) repo, ros2 branch)
* naoqi_bridge_msgs ([naoqi_bridge_msgs2](https://github.com/ros-naoqi/naoqi_bridge_msgs2) repo)
* naoqi_driver ([naoqi_driver2](https://github.com/ros-naoqi/naoqi_driver2) repo)

Additionally, the pepper_meshes and nao_meshes have been ported ([pepper_meshes2](https://github.com/ros-naoqi/pepper_meshes2) and [nao_meshes2](https://github.com/ros-naoqi/nao_meshes2) repos).

The packages are being released, the meshes packages aren't released yet but will hopefully be released soon (you can always build them from source in the meantime). The targeted ROS2 distros are __Foxy__ and __Galactic__ for now.

closed 2022-09-15 by mbusy