ros-naoqi/naoqi_driver #144: unable to install naoqi sdk

UdayRockzz, 2021-08-24:
I am following this [page](http://wiki.ros.org/nao/Tutorials/Installation) installing naoqi SDK. While executing this command  `$` ~/naoqi/naoqi-sdk-2.5.7.1-linux64/naoqi   
it stuck in between
[image]
what could be the problem? I am using ubuntu 16.04. ROS Kinetic.
(https://user-images.githubusercontent.com/39263163/130634107-88002250-789b-452c-980e-b963425287f0.png)

mbusy, 2021-09-01:
That tutorial is outdated... I assume that ultimately you want to be able to launch the driver and not install the naoqi SDK ?

To install the driver in kinetic you basically need to install the following packages:
> sudo apt-get install ros-kinetic-pepper-meshes
> sudo apt-get install ros-kinetic-naoqi-libqi
> sudo apt-get install ros-distribution-naoqi-libqicore
> sudo apt-get install ros-kinetic-naoqi-driver
> sudo apt-get install ros-kinetic-naoqi-bridge-msgs

Then you can use a roslaunch command to launch the driver for your robot

UdayRockzz, 2021-09-20:
Thanks for the reply @mbusy. yeah, I have installed all the above packages which you have mentioned. and I am following this  [tutorial](http://wiki.ros.org/pepper/Tutorials) for connecting pepper robot physically. when I have entered this commend 

> $ roslaunch pepper_bringup pepper_full_py.launch nao_ip:=<yourRobotIP> roscore_ip:=<roscore_ip>   python version 

I am getting this issue.  how to solve it? 
[image]

and is there any documentation to follow to connect with the pepper robot(not virtually)? Thanks in advance.

mbusy, 2021-09-23:
What I would advise would be to try and use naoqi_driver's launchfile instead of pepper_bringup's (pepper_bringup's launchfile should be working for kinetic though, but let's do that one step at a time).

The launch command should be:
```shell
# You can get your network interface by typing the ifconfig command in your console 
roslaunch naoqi_driver naoqi_driver.launch nao_ip:=your_robot_ip network_interface:=your_interface
```

UdayRockzz, 2021-09-24:
well, I tired this command 

 `roslaunch naoqi_driver naoqi_driver.launch nao_ip:=your_robot_ip network_interface:=your_interface` , and it worked and the final output is as below 
[image]

and when I run `rosrun rviz rviz` command in another shell and I added Robotmodel. the output is like below 
[image]  

As it is mentioned in this[ tutorial ](http://wiki.ros.org/pepper/Tutorials) in section 2.4

> #We provide a pre-configured RViz configuration which has all basic displays such as TF, RobotModel, Camera, Lasers and Sonars. This configuration can be found inside your catkin workspace under
 `src/pepper_robot/pepper_description/config/pepper.rviz`

mbusy, 2021-09-24:
I remember encountering a similar issue, it might be because of the "map" fixed frame in your RViz config. Since the frame does not exist, the other frames can't be properly placed. Try selecting odom or base_footprint instead

UdayRockzz, 2021-09-24:
Thanks alot. it worked when I changed fixed frame to "Odom" and when I add camera and image options, the output is like below. 
[image] . how to view camera and image?

mbusy, 2021-09-24:
You can ask RViz to display an image. Yiu just have to specify the topic on which that image is published, so /naoqi_driver/camera/front/image_raw for instance (you should have a list of proposed topics to choose from)

UdayRockzz, 2021-09-24:
yeah, I found it. The camera and image options topics are working only before selecting a fixed frame to Odom. it is not working after selecting the fixed frame to Odom.

dac31415, 2021-09-24:
Hi @UdayRockzz, we might be having similar problems. Are you using a virtual machine? I tried the driver by itself as recommend it here by @mbusy and is working for me; laser (as a whole) and the sonars. However, for the cameras, I'm having this message. 
`front_cameraCamera` Handle is empty - cannot retrieve image
`front_cameraMight` be a NAOqi problem. Try to restart the ALVideoDevice.
 For all my cameras. I'm using ROS Melodic in Ubuntu 18.04 in VMware.  I'm going to try now teleop using this cause when I tried pepper_bringup, I couldn't make it work either. Full no responses at all and full_py crashed.

mbusy, 2021-09-24:
@UdayRockzz What do you mean by not working, no image is published after selecting the odom frame?

UdayRockzz, 2021-09-30:
sorry. it was a mistake from my side, I am able to select the topics properly. and I have another problem here. when I run the pepper with teleop_twist_keyboard  command. sometimes it is working properly and sometimes it is not. For example,  when I press the forward(I) key, it is not moving forward sometimes but turning left/right.   and sometimes even when I press any command repeatedly it is not even moving(idle/inactive). and sometimes it is turning around and talking without pressing any key. because of this problem, I am unable to take proper navigation using slam_gmapping. And,  I have attached the video [link](https://drive.google.com/file/d/15CqI7Zk2RNA2XeiNtDOWFaDATBvIXKpT/view?usp=sharingl) here
but it's too long. sorry for that.  but how should I get rid of it? what could be the problem? Is this because of an error in pepper or API or in teleop command or something else?

UdayRockzz, 2021-09-30:
> Hi @UdayRockzz, we might be having similar problems. Are you using a virtual machine? I tried the driver by itself as recommend it here by @mbusy and is working for me; laser (as a whole) and the [...]

UdayRockzz, 2021-09-30:
@dac31415 I have just seen your issue. sorry for the late reply.  I am not using VM. I am doing it on Ubuntu 16.04 Ros; Kinetic on real pepper robot.

UdayRockzz, 2021-09-30:
@mbusy sorry. it was a mistake from my side, I am able to select the topics properly. and I have another problem here. when I run the pepper with teleop_twist_keyboard  command. sometimes it is working properly and sometimes it is not. For example,  when I press the forward(I) key, it is not moving forward sometimes but turning left/right.   and sometimes even when I press any command repeatedly it is not even moving(idle/inactive). and sometimes it is turning around and talking without pressing any key. because of this problem, I am unable to take proper navigation using slam_gmapping. And,  I have attached the video [link](https://drive.google.com/file/d/15CqI7Zk2RNA2XeiNtDOWFaDATBvIXKpT/view?usp=sharingl) here
but it's too long. sorry for that.  but how should I get rid of it? what could be the problem? Is this because of an error in pepper or API or in teleop command or something else?

dac31415, 2021-10-01:
@UdayRockzz no worries. I was able to fix my issue during my weekend. I'm also using the real robot but I have windows, so I use VMware and at first, I thought that was an issue.

UdayRockzz, 2021-10-01:
@dac31415 That was great. BTW, is your robot moving correctly when you are using teleop command? I am facing some issues while using teleop command, as you can see  [here](https://drive.google.com/file/d/15CqI7Zk2RNA2XeiNtDOWFaDATBvIXKpT/view). Are you facing  a similar kind of issue?

dac31415, 2021-10-01:
@UdayRockzz sometimes, but is not really a problem for me. The only thing I did is reducing the velocity.

mbusy, 2022-09-15:
Since the original issue has been resolved, I'll close it

closed 2022-09-15 by mbusy