ros-naoqi/naoqi_driver #145: No display of cloud points. Topic is empty.

dac31415, 2021-09-28:
Hello, I've been having this issue for a while. I can't display the cameras and therefore no display of points required for mapping. 
I launch the naoqi driver as follows:

> roslaunch naoqi_driver naoqi_driver.launch nao_ip:=192.168.8.90 network_interface:=ens33

and have access to the cameras topics and can visualize them (raw images). 
I then follow the steps mentioned in [here ](https://groups.google.com/g/ros-sig-aldebaran/c/cwqfb5aLlmc) to map into the new topics. 

FYI, I uncommented the lines required for the depth in  camera_info_definitions.hpp but an error appeared after doing catkin_make. So, instead I used the file from the forked version of Softbank robotics. 

After doing the remapping of topics and try to subscribe to the camera (rviz), the following message appears. 

<pre>get_language is resetting
get_language reset
<font color="#FCE94F"><b>naoqi_driver initialized</b></font>
front_cameraCamera Handle is empty - cannot retrieve image
front_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
front_cameraCamera Handle is empty - cannot retrieve image
front_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
front_cameraCamera Handle is empty - cannot retrieve image
front_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
front_cameraCamera Handle is empty - cannot retrieve image
front_cameraMight be a NAOqi problem. Try to restart the ALVideoDevice.
</pre>

I'm using ROS Melodic on Ubuntu 18.04. I checked, and all dependencies have been installed. 

Thank you.

dac31415, 2021-09-29:
Update: the ALVideoDevice is working now. Rebooting the robot solved this. However, still no cloudpoints.

dac31415, 2021-09-30:
Finally, I was able to use to have access to the depth clouds with the Full bring up. The forked version work just fine.

closed 2021-09-30 by dac31415

UdayRockzz, 2021-11-25:
@dac31415 which forked version have you used? could you please share the link. 
TIA

dac31415, 2021-11-26:
@UdayRockzz change the camera header file from the aldebaran github.

UdayRockzz, 2021-12-01:
@dac31415 Thanks a lot for responding. I have downloaded and followed all those steps which @mbusy suggested but still, I am unable to get the point cloud information. If it is possible could you please share the file with me (uday038@gmail.com)? It will be helpful for my analysis.