ros-naoqi/naoqi_driver #146: naoqi_driver is not publishing point cloud

UdayRockzz, 2021-12-01:
Hello,

I am working on octomap package where I need point cloud of the pepper robot.  unfortunately naoqi_driver is not publishing point cloud. and these are my rostopic list: 

/clock
/cmd_vel
/diagnostics
/joint_angles
/joint_states
/move_base_simple/goal
/naoqi_driver/audio
/naoqi_driver/bumper
/naoqi_driver/camera/bottom/camera_info
/naoqi_driver/camera/bottom/image_raw
/naoqi_driver/camera/bottom/image_raw/compressed
/naoqi_driver/camera/bottom/image_raw/compressed/parameter_descriptions
/naoqi_driver/camera/bottom/image_raw/compressed/parameter_updates
/naoqi_driver/camera/bottom/image_raw/theora
/naoqi_driver/camera/bottom/image_raw/theora/parameter_descriptions
/naoqi_driver/camera/bottom/image_raw/theora/parameter_updates
/naoqi_driver/camera/depth/camera_info
/naoqi_driver/camera/depth/image_raw
/naoqi_driver/camera/depth/image_raw/compressed
/naoqi_driver/camera/depth/image_raw/compressed/parameter_descriptions
/naoqi_driver/camera/depth/image_raw/compressed/parameter_updates
/naoqi_driver/camera/depth/image_raw/compressedDepth
/naoqi_driver/camera/depth/image_raw/compressedDepth/parameter_descriptions
/naoqi_driver/camera/depth/image_raw/compressedDepth/parameter_updates
/naoqi_driver/camera/depth/image_raw/theora
/naoqi_driver/camera/depth/image_raw/theora/parameter_descriptions
/naoqi_driver/camera/depth/image_raw/theora/parameter_updates
/naoqi_driver/camera/front/camera_info
/naoqi_driver/camera/front/image_raw
/naoqi_driver/camera/front/image_raw/compressed
/naoqi_driver/camera/front/image_raw/compressed/parameter_descriptions
/naoqi_driver/camera/front/image_raw/compressed/parameter_updates
/naoqi_driver/camera/front/image_raw/theora
/naoqi_driver/camera/front/image_raw/theora/parameter_descriptions
/naoqi_driver/camera/front/image_raw/theora/parameter_updates
/naoqi_driver/camera/stereo/camera_info
/naoqi_driver/camera/stereo/image_raw
/naoqi_driver/camera/stereo/image_raw/compressed
/naoqi_driver/camera/stereo/image_raw/compressed/parameter_descriptions
/naoqi_driver/camera/stereo/image_raw/compressed/parameter_updates
/naoqi_driver/camera/stereo/image_raw/theora
/naoqi_driver/camera/stereo/image_raw/theora/parameter_descriptions
/naoqi_driver/camera/stereo/image_raw/theora/parameter_updates
/naoqi_driver/hand_touch
/naoqi_driver/head_touch
/naoqi_driver/imu/base
/naoqi_driver/imu/torso
/naoqi_driver/info
/naoqi_driver/laser
/naoqi_driver/odom
/naoqi_driver/sonar/back
/naoqi_driver/sonar/front
/naoqi_driver_node/audio
/naoqi_driver_node/bumper
/naoqi_driver_node/camera/bottom/camera_info
/naoqi_driver_node/camera/bottom/image_raw
/naoqi_driver_node/camera/bottom/image_raw/compressed
/naoqi_driver_node/camera/bottom/image_raw/compressed/parameter_descriptions
/naoqi_driver_node/camera/bottom/image_raw/compressed/parameter_updates
/naoqi_driver_node/camera/bottom/image_raw/theora
/naoqi_driver_node/camera/bottom/image_raw/theora/parameter_descriptions
/naoqi_driver_node/camera/bottom/image_raw/theora/parameter_updates
/naoqi_driver_node/camera/depth/camera_info
/naoqi_driver_node/camera/depth/image_raw
/naoqi_driver_node/camera/depth/image_raw/compressed
/naoqi_driver_node/camera/depth/image_raw/compressed/parameter_descriptions
[... 61 lines cut ...]
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/boost/function/detail/prologue.hpp:13,
                 from /usr/include/boost/function.hpp:24,
                 from /opt/ros/kinetic/include/qi/anyfunction.hpp:11,
                 from /opt/ros/kinetic/include/qi/signal.hpp:12,
                 from /opt/ros/kinetic/include/qi/anyobject.hpp:15,
                 from /home/uday/Desktop/pepper/workspace/catkin_ws/src/naoqi_driver/include/naoqi_driver/tools.hpp:30,
                 from /home/uday/Desktop/pepper/workspace/catkin_ws/src/naoqi_driver/src/converters/converter_base.hpp:24,
                 from /home/uday/Desktop/pepper/workspace/catkin_ws/src/naoqi_driver/src/converters/camera.hpp:24,
                 from /home/uday/Desktop/pepper/workspace/catkin_ws/src/naoqi_driver/src/converters/camera.cpp:21:
/usr/include/c++/5/bits/stl_vector.h:448:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::vector<_Tp, _Alloc>&&) [with _Tp = double; _Alloc = std::allocator<double>]
       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
       ^
/usr/include/c++/5/bits/stl_vector.h:470:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = double; _Alloc = std::allocator<double>]
       operator=(initializer_list<value_type> __l)
       ^
naoqi_driver/CMakeFiles/naoqi_driver.dir/build.make:230: recipe for target 'naoqi_driver/CMakeFiles/naoqi_driver.dir/src/converters/camera.cpp.o' failed
make[2]: *** [naoqi_driver/CMakeFiles/naoqi_driver.dir/src/converters/camera.cpp.o] Error 1
CMakeFiles/Makefile2:8397: recipe for target 'naoqi_driver/CMakeFiles/naoqi_driver.dir/all' failed
make[1]: *** [naoqi_driver/CMakeFiles/naoqi_driver.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

and even I have checked with SoftBank's [version.](https://github.com/softbankrobotics-research/naoqi_driver) even though I am not getting point cloud information. Does anyone solve this issue? or I might be missing any thing?

sasilva1998, 2022-02-27:
Hello, could somebody solve this issue? I am using the Pepper Robot as well but I dont get any message in the depth points camera topic

UdayRockzz, 2022-03-07:
Hi @sasilva1998, try this with this [repo](https://github.com/ros-naoqi/pepper_robot). You will get depth information.

sasilva1998, 2022-03-08:
At last I got it working by using the C++ sdk, however, I get a really low frequency. I get about 1 pointcloud message per second. How much frequency do you get from the Depth Camera @UdayRockzz?

Thanks for your advice

UdayRockzz, 2022-03-10:
oh. you are getting point cloud data from C++ SDK. that's nice. As I mentioned before, I am using python based sdk and this is the rosbag from the topic
. even same for me. 

[image]