ros-naoqi/naoqi_driver #147: ROS in (virtual) NAOV6  simulation

ysys98, 2022-03-09:
Hello, I haven't started school yet, so I don't have a physical NAO around me. How can I connect a virtual NAO on my computer and conduct simulation just like connecting a physical NAO？

I follow these to try connect a virtual NAO
https://github.com/ros-naoqi/naoqi_driver/issues/134
https://github.com/ros-naoqi/naoqi_driver/issues/144
http://wiki.ros.org/nao/Tutorials/Getting-Started

I ran the following instructions as recommended by the authorities in ROS wiki 
[image]

And then I follow #134 #144 run ：
roslaunch naoqi_driver naoqi_driver.launch nao_ip:=127.0.0.1 network_interface:=enp0s3

But I got an error：
[image]
it said ：Receiving information about robot model
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  Cannot find service 'ALMemory' in index

So what should I do next？

mbusy, 2022-03-09:
Hi @ysys98, did you take a look at [that repo](https://github.com/softbankrobotics-research/qibullet) ? More specifically the [ROS wiki page](https://github.com/softbankrobotics-research/qibullet/wiki/Tutorials:-ROS-Interface) and the [ROS example script](https://github.com/softbankrobotics-research/qibullet/blob/master/examples/robot_ros_test.py). That might help you simulate a NAO and connecting it to the driver

ysys98, 2022-03-17:
> 嗨@ysys98，你看过[那个回购](https://github.com/softbankrobotics-research/qibullet)了吗？更具体地说，[ROS wiki 页面](https://github.com/softbankrobotics-research/qibullet/wiki/Tutorials:-ROS-Interface)和[ROS 示例脚本](https://github.com/softbankrobotics-research/qibullet/blob/master/examples/robot_ros_test.py)。这可能会帮助您模拟 NAO 并将其连接到驱动程序

Thank you！ @mbusy 
Thank you and I've done it this way
But I have some questions,
When I run some examples according to the qibullet tutorial,
[image]

Then I started rviz
But I didn't find Odom in the fixed frame

Another question I have is that when I simulate a Nao in qibullet, he is standing in pybullet, but when I open rviz, he is half embedded in the map, that is, he has no base_ Footprint, why? How can I solve these two problems
[image]
[image]

[image]

1. I can't find Odom in the fixed frame
2. There is no base in the fixed frame in rviz_ footprint

mbusy, 2022-03-17:
You're welcome! If you launch the ROS example, the odom frame should appear in RViz: the qiBullet ROS wrapper is used and publishes the odom frame for all virtual robots ([this method](https://github.com/softbankrobotics-research/qibullet/blob/486e5943a9886a777eeacdc06e97e323ccd0cc31/qibullet/ros_wrapper.py#L173) is used). If you launch a classical example, (like [this one](https://github.com/softbankrobotics-research/qibullet/blob/master/examples/light_position.py) for instance) the ROS wrapper isn't used and no extra frames will be published. To be clear, qiBullet doesn't launch naoqi_driver, but uses its own ROS wrapper to emulate naoqi_driver's functionalities.

To be even more precise: real robot + naoqi_driver = simulated qibullet robot + qibullet ROS wrapper. In terms of topics and frames, the usage should be transparent. So make sure that a ROS wrapper is launched, like in the [qibullet ROS example](https://github.com/softbankrobotics-research/qibullet/blob/master/examples/robot_ros_test.py).

Regarding your second problem, solve your first problem and then use odom as a fixed frame in RViz. The robot won't be embedded in the ground anymore. (The base link of NAO is located in its torso, that's why it appears embedded in the ground)

Since this issue is starting to discuss content related to [qiBullet](https://github.com/softbankrobotics-research/qibullet) and not naoqi_driver, I'll close this issue. If you need any additional information on that specific issue, you can of course open an [issue](https://github.com/softbankrobotics-research/qibullet/issues) or a [discussion](https://github.com/softbankrobotics-research/qibullet/discussions) in the qiBullet repo, and we'll continue the discussion there!