ros-naoqi/naoqi_driver #150: Noetic naoqi_driver.launch Can't find service ROS-Driver-Audio

LaiYanKai, 2022-08-18:
Hi,
I have tried to use naoqi_driver.launch on ROS1-Noetic by compiling from source (#138):
```
mkdir -p naoqi_ws/src
cd naoqi_ws/src
git clone https://github.com/ros-naoqi/libqi.git
cd libqi
git checkout noetic-devel
cd ..
git clone https://github.com/ros-naoqi/libqicore.git
cd libqicore
git checkout noetic-devel
cd ..
git clone https://github.com/ros-naoqi/naoqi_bridge_msgs.git
git clone https://github.com/ros-naoqi/naoqi_driver.git
cd ..
catkin_make
```
The steps work. Currently, libqi and libqicore have been developed for noetic, but not completely so for naoqi_bridge_msgs and naoqi_driver.

When launching naoqi_driver.launch, the ***Can't find service*** error pops up:
```
roslaunch naoqi_driver naoqi_driver.launch  nao_ip:=192.168.218.83 roscore_ip:=192.168.218.190 network_interface:=enp0s3
... logging to /home/mallab/.ros/log/05d57566-1ec3-11ed-8a1a-7b182742d2bf/roslaunch-mallab-VirtualBox-3200.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mallab-VirtualBox:39239/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [3208]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 05d57566-1ec3-11ed-8a1a-7b182742d2bf
process[rosout-1]: started with pid [3218]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [3221]
Receiving information about robot model
Robot detected: NAO V6.0
set prefix successfully to naoqi_driver
using ip address: 192.168.218.190 @ enp0s3
found a catkin prefix /home/mallab/naoqi_ws/src/naoqi_driver/share/boot_config.json
load boot config from /home/mallab/naoqi_ws/src/naoqi_driver/share/boot_config.json
found a catkin URDF /home/mallab/naoqi_ws/src/naoqi_driver/share/urdf/nao.urdf
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  	ALProxy::ALProxy
	Can't find service: ROS-Driver-Audio
================================================================================REQUIRED process [naoqi_driver-2] has died!
process has died [pid 3221, exit code -6, cmd /home/mallab/naoqi_ws/devel/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://192.168.218.83:9559 --roscore_ip=192.168.218.190 --network_interface=enp0s3 --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/mallab/.ros/log/05d57566-1ec3-11ed-8a1a-7b182742d2bf/naoqi_driver-2.log].
log file: /home/mallab/.ros/log/05d57566-1ec3-11ed-8a1a-7b182742d2bf/naoqi_driver-2*.log
Initiating shutdown!
================================================================================
[naoqi_driver-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I'm using VirtualBox on Windows 11 (host). The VirtualBox is running Ubuntu 20.04. Referencing #96, I've made sure that **Bridged networking** is used. I am able to SSH to nao (and using Robot settings) from both the host and Ubuntu, and ping to VirtualBox's Ubuntu while in nao's SSH. The launch also detected the robot. I don't think it is an issue like #96. What could be the problem?

mbusy, 2022-08-20:
Hi @LaiYanKai, unfortunately I think it's an error like #96, I had the same problem while using the driver for Noetic, even for ROS2 (I'll have to those changes as well at some point). I remember tracking the source of the issue, I don't remember it's precise source but it's much more lower than the driver, it might be a NAOqi related issue. Can you try to disable the audio ([here](https://github.com/ros-naoqi/naoqi_driver/blob/460e778b60e48432aecb506e303db8ce64c8315e/share/boot_config.json#L85))? Setting the parameter to false should work (the audio part of the driver will be disabled though)

LaiYanKai, 2022-08-22:
Hi, I will get back to you within a day

LaiYanKai, 2022-08-23:
Works. The following issues showed up:
### Conversion Failure in IMU.cpp
I have no problems echoing the IMU and getting the messages. However, this exception message shows up repeatedly:
```Conversion from v(void) to f(float) failed => set to -1```. 

Inserting ```std::cout << i <<std::endl``` [before line 192 of from_any_value.cpp](https://github.com/ros-naoqi/naoqi_driver/blob/master/src/tools/from_any_value.cpp#L192) prints out:
```
0
Conversion from v(void) to f(float) failed => set to -1
1
2
3
4
5
6
7
8
9
```
Changing from ```Device/SubDeviceList/InertialSensorBase/```... to ```Device/SubDeviceList/InertialSensor/```...  in 
 [imu.cpp](https://github.com/ros-naoqi/naoqi_driver/blob/master/src/converters/imu.cpp) does not remove the message. Current SDK uses the latter as described [here](https://developer.softbankrobotics.com/nao6/nao-documentation/nao-developer-guide/actuator-sensor-list#inertial). I don't think the error is caused by this.

###  Repeating No-transform Error Messages:
While running `naoqi_driver.launch`, the `ROS_ERROR` keeps showing up:
```Do not calculate NAO Footprint, no transform possible```

This error appears after the line ```tf2_buffer->canTransform("odom", "l_sole", time, ros::Duration(0.1) )``` in [nao_footprint.hpp](https://github.com/ros-naoqi/naoqi_driver/blob/master/src/converters/nao_footprint.hpp#L42), when no transform is possible. However, there appears to be a way to transform between `l_sole` and `odom` in the tf tree in
[frames.pdf](https://github.com/ros-naoqi/naoqi_driver/files/9399223/frames.pdf). The pdf is generated using:
```
rosrun tf2_tools view_frames.py
evince frames.pdf
```

### Low frame rate for Camera
The camera images are transporting correctly. The frame is around 1 to 2 fps in the default setting. Changing the `fps` in [boot_config.json](https://github.com/ros-naoqi/naoqi_driver/blob/master/share/boot_config.json#L9) to 30 improves the rate to about 3fps, but it is still quite slow... 

Changing the `resolution` does not appear to change the resolution or improve the frame rate, as the `...\camera_info` topic still reports 320px by 480px, unless there are specific resolution values? (I tried 0.5).

In your experience, is at least 20 FPS on 320x480px possible from previous ROS versions?

LaiYanKai, 2022-08-23:
Hi,
Publishing goal from Rviz works as well... the robot tries to move to the intended target.

I will try to simulate NAO in Gazebo within a week by trying nao_virtual... will report any problems here...

mbusy, 2022-09-05:
Alright. I'll keep the issue opened regarding the ROS-Driver-Audio problem. If I have the time to investigate and if it is indeed a problem unrelated to the driver, I might close it, or add a workaround. Regarding the `Repeating No-transform Error Messages`, #120 is already opened. Regarding the `Low frame rate for Camera`, I believe that an issue is already opened too, it's most likely a network related problem. If it's not the case, I'll open a separate issue. Finally regarding the `Conversion Failure in IMU.cpp`, it is somwhat mentioned in #117, It's worth investigating a bit more. I unfortunately don't have access to a NAO to investigate however

LaiYanKai, 2022-09-06:
Hi,
I've forgotten to update.

I've printed out the ROS time on Lines 273 and 298 in 
[camera.cpp](https://github.com/ros-naoqi/naoqi_driver/blob/460e778b60e48432aecb506e303db8ce64c8315e/src/converters/camera.cpp#L273). The whole thing lasts 20ms, but something else is limiting the frequency of this function to around 200-400ms. (I've used QVGA by setting resolution to 0). 

Robot Settings / Monitor has no problem with the frame rate, it is significantly faster.

Could I kindly check where the containing function callAll is called? I know it is called as part of a thread / shared pointer, but it'll be better if there are some hints as to the bottleneck... I may be able to figure out the bottleneck of the frame rate from there...

I've tried the python sdk, but it is quite problematic as it uses 2.7 and I need low latency. The cpp SDK qibuild messes with the ROS catkin_make. So this is the best choice.