ros-naoqi/naoqi_driver #158: "Can't find service: ROS-Driver-Audio" - ROS Noetic Ubuntu Ubuntu 20.04 - Pepper 1.8

johnnypipey, 2023-06-16:
I am trying to connect a Pepper 1.8 robot to ROS Noetic aiming for use in research and for autonomous navigation tasks such as a tour guide. I have followed the tutorials and steps to install the ROS driver however i am met with the ROS-Driver-Audio Error whenever I run the roslaunch naoqi_driver.
Please see terminal Output:

 roslaunch naoqi_driver naoqi_driver.launch nao_ip:=10.39.23.56
... logging to /home/johnpipe/.ros/log/2195fa48-0c50-11ee-a0e5-8965157f7919/roslaunch-galatea-53004.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://galatea:45307/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [53019]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2195fa48-0c50-11ee-a0e5-8965157f7919
process[rosout-1]: started with pid [53036]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [53039]
Receiving information about robot model
Robot detected/NAOqi version: Pepper 1.8 / 2.5.10.7
set prefix successfully to naoqi_driver
using ip address: 127.0.0.1 @ eth0
found a catkin prefix /opt/ros/noetic/share/naoqi_driver/share/boot_config.json
load boot config from /opt/ros/noetic/share/naoqi_driver/share/boot_config.json
found a catkin URDF /opt/ros/noetic/share/naoqi_driver/share/urdf/pepper.urdf
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  	ALProxy::ALProxy
	Can't find service: ROS-Driver-Audio
================================================================================REQUIRED process [naoqi_driver-2] has died!
process has died [pid 53039, exit code -6, cmd /opt/ros/noetic/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://10.39.23.56:9559 --nao_ip=10.39.23.56 --nao_port=9559 --user=nao --password=no_password --roscore_ip=127.0.0.1 --network_interface=eth0 --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/johnpipe/.ros/log/2195fa48-0c50-11ee-a0e5-8965157f7919/naoqi_driver-2.log].
log file: /home/johnpipe/.ros/log/2195fa48-0c50-11ee-a0e5-8965157f7919/naoqi_driver-2*.log
Initiating shutdown!
================================================================================
[naoqi_driver-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Hardrick98, 2023-06-28:
Maybe it's the Naoqi version? I'm encountering the same issue and I have naoqi 2.5.8...

victorpaleologue, 2023-07-26:
2.5 should be alright. Do you need the audio driver? I think there's a way to disable audio.