ros-naoqi/naoqi_driver #161: undefined symbol: _ZN2qi6detail22throwConversionFailureEPNS_13TypeInterfaceES2_RKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE

brain0071, 2023-11-15:
I want to run a naoqi robot simulation.
I downloaded ```naoqi-sdk-2.8.5.10-linux64```, follow this tutorial:"https://wiki.ros.org/nao/Tutorials/Installation", then run this comman in terminal:```./naoqi --verbose --broker-ip 127.0.0.1```
Then I clone this repo and compiled it use "catkin_make", 
When I run this command: ```roslaunch naoqi_driver naoqi_driver.launch nao_ip:=127.0.0.1```, I got this Error.
How should I solve it?

```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://naodai-ThinkPad-X13-Gen-1:44067/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

process[naoqi_driver-1]: started with pid [30688]
/home/naodai/WorkSpace/naoqi_ws/devel/lib/naoqi_driver/naoqi_driver_node: symbol lookup error: /home/naodai/WorkSpace/naoqi_ws/devel/lib/naoqi_driver/naoqi_driver_node: undefined symbol: _ZN2qi6detail22throwConversionFailureEPNS_13TypeInterfaceES2_RKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
================================================================================REQUIRED process [naoqi_driver-1] has died!
process has died [pid 30688, exit code 127, cmd /home/naodai/WorkSpace/naoqi_ws/devel/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://127.0.0.1:9559 --nao_ip=127.0.0.1 --nao_port=9559 --user=nao --password=no_password --roscore_ip=127.0.0.1 --network_interface=eth0 --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/naodai/.ros/log/cfe0a1c2-837c-11ee-8cb7-046c5939ab25/naoqi_driver-1.log].
log file: /home/naodai/.ros/log/cfe0a1c2-837c-11ee-8cb7-046c5939ab25/naoqi_driver-1*.log
Initiating shutdown!
================================================================================
[naoqi_driver-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

victorpaleologue, 2023-11-15:
Melodic is a bit old, so I am not sure naoqi_driver from the sources is compatible with melodic's libqi.
You should check out libqi too in your sources (https://github.com/ros-naoqi/libqi, branch  `ros`), and libqicore (https://github.com/ros-naoqi/libqicore, branch  `ros`), as stated in the README: https://github.com/ros-naoqi/naoqi_driver#dependencies

brain0071, 2023-11-15:
So if I use Noetic, it is better?

brain0071, 2023-11-15:
> Melodic is a bit old, so I am not sure naoqi_driver from the sources is compatible with melodic's libqi. You should check out libqi too in your sources (https://github.com/ros-naoqi/libqi, branch [...]

So if I use Noetic, it is better?

brain0071, 2023-11-15:
> Yes

Ok,thanks a lot!

brain0071, 2023-11-16:
> Yes

Sorry, Can you provide a link to naoqi sdk 2.9? I can't find it. I am trying using Noetic.

victorpaleologue, 2023-11-16:
There is no NAOqi SDK for 2.9, it only works with real robots. Hence the tweak I suggested.