ros-naoqi/naoqi_driver #162: `Connect error: 111: Connection refused` (Pepper with NAOqi 2.9 + ROS noetic)

kochigami, 2024-01-22:
Hi, 

I am trying to use the master branch of naoqi_driver with Pepper using NAOqi 2.9.3.
My ROS version is noetic.

First, I set NAO_IP and ROS_IP as environment variables, and connected Pepper and PC to the same network.

Then, I executed the below command:
```
roslaunch naoqi_driver naoqi_driver.launch network_interface:=<interface> password:=<password>
```
Finally, I got the error message `Connect error: 111: Connection refused` as shown below.
```
... logging to /home/kochigami/.ros/log/50b6728e-b8ef-11ee-b3fa-51fdea7b95cc/roslaunch-kochigami-ThinkPad-P16-Gen-1-5527.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.8.240:36281/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [5558]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 50b6728e-b8ef-11ee-b3fa-51fdea7b95cc
process[rosout-1]: started with pid [5591]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [5594]
Connect error: 111: Connection refused
================================================================================REQUIRED process [naoqi_driver-2] has died!
process has finished cleanly
log file: /home/kochigami/.ros/log/50b6728e-b8ef-11ee-b3fa-51fdea7b95cc/naoqi_driver-2*.log
Initiating shutdown!
================================================================================
[naoqi_driver-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I could do `ssh nao@<ip>`.

Is there anyone who gets the same error?
Please let me know if you know of a solution.

P.S.

When I tried naoqi_driver (master branch) with ROS noetic and Pepper using NAOqi 2.5.11.13, it worked successfully.
```
roslaunch naoqi_driver naoqi_driver.launch network_interface:=<interface>
```
```
... logging to /home/kochigami/.ros/log/ddd88412-b8f0-11ee-b3fa-51fdea7b95cc/roslaunch-kochigami-ThinkPad-P16-Gen-1-6168.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.8.240:45045/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [6199]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ddd88412-b8f0-11ee-b3fa-51fdea7b95cc
process[rosout-1]: started with pid [6232]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [6235]
Receiving information about robot model
Robot detected/NAOqi version: Pepper 1.8 / 2.5.11.13
set prefix successfully to naoqi_driver
using ip address: 127.0.0.1 @ <interface>
found a catkin prefix /home/kochigami/catkin_ws/src/naoqi_driver/share/boot_config.json
load boot config from /home/kochigami/catkin_ws/src/naoqi_driver/share/boot_config.json
found a catkin URDF /home/kochigami/catkin_ws/src/naoqi_driver/share/urdf/pepper.urdf
RightBumperPressed
LeftBumperPressed
BackBumperPressed
ROS-Driver-RightBumperPressed : Start
HandRightBackTouched
HandRightLeftTouched
HandRightRightTouched
HandLeftBackTouched
HandLeftLeftTouched
HandLeftRightTouched
ROS-Driver-HandRightBackTouched : Start
FrontTactilTouched
MiddleTactilTouched
RearTactilTouched
ROS-Driver-FrontTactilTouched : Start
registered subscriber:	teleop
registered subscriber:	moveto
registered subscriber:	speech
nodehandle reset 
using master ip: http://127.0.0.1:11311
NOT going to re-register the converters
camera/bottom/image_raw is resetting
camera/bottom/image_raw reset
camera/depth/image_raw is resetting
camera/depth/image_raw reset
/diagnostics is resetting
/diagnostics reset
camera/front/image_raw is resetting
camera/front/image_raw reset
imu/base is resetting
imu/base reset
imu/torso is resetting
imu/torso reset
info is resetting
load robot description from file
info reset
/joint_states is resetting
/joint_states reset
laser is resetting
laser reset
odom is resetting
odom reset
sonar is resetting
sonar reset
camera/stereo/image_raw is resetting
camera/stereo/image_raw reset
teleop is resetting
teleop reset
moveto is resetting
moveto reset
speech is resetting
speech reset
get_robot_config is resetting
get_robot_config reset
set_language is resetting
set_language reset
get_language is resetting
get_language reset
naoqi_driver initialized
```

victorpaleologue, 2024-01-22:
Did you set the port to 9443?

If yes and it still does not work, does it work better when you pass the connection information this way?
```bash
naoqi_driver.launch nao_ip:=<ip> nao_port:=<port> 
```

And does it work better if you prefix the IP with "tcps://"?

Let me know what works and I'll update the README accordingly. PRs are welcome too ;-)

kochigami, 2024-01-23:
Thank you very much for your advice!!
Unfortunately, it seems that it did not succeed...

1. When I executed 
`roslaunch naoqi_driver naoqi_driver.launch network_interface:=<interface> nao_port:=9443 password:=<password>`
and 
`roslaunch naoqi_driver naoqi_driver.launch nao_ip:=10.232.44.27 nao_port:=9443 password:=<password> network_interface:=<interface>`
and
`roslaunch naoqi_driver naoqi_driver.launch nao_ip:=10.232.44.27 nao_port:=9443 password:=<password>`

The error `terminate called after throwing an instance of 'qi::FutureUserException'
  what():  Cannot find service 'ALMemory' in index` appeared as follows.
When I did ssh to Pepper, `qicli info ALMemory` returned the results.

```
... logging to /home/kochigami/.ros/log/f412413a-b992-11ee-89c2-91b4da16ac22/roslaunch-kochigami-ThinkPad-P16-Gen-1-7195.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.232.44.17:33063/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [7226]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f412413a-b992-11ee-89c2-91b4da16ac22
process[rosout-1]: started with pid [7259]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [7262]
Receiving information about robot model
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  Cannot find service 'ALMemory' in index
================================================================================REQUIRED process [naoqi_driver-2] has died!
process has died [pid 7262, exit code -6, cmd /home/kochigami/catkin_ws/devel/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://10.232.44.27:9443 --nao_ip=10.232.44.27 --nao_port=9443 --user=nao --password=<password> --roscore_ip=127.0.0.1 --network_interface=<interface> --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/kochigami/.ros/log/f412413a-b992-11ee-89c2-91b4da16ac22/naoqi_driver-2.log].
log file: /home/kochigami/.ros/log/f412413a-b992-11ee-89c2-91b4da16ac22/naoqi_driver-2*.log
Initiating shutdown!
================================================================================
[naoqi_driver-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

2. Also, when I tried
`roslaunch naoqi_driver naoqi_driver.launch nao_ip:=10.232.44.27 nao_port:=9443`
and
`roslaunch naoqi_driver naoqi_driver.launch nao_ip:=10.232.44.27 nao_port:=9443 network_interface:=<interface>`,

I got the `disconnected` error message as follows. So maybe I need password as an argument.

```
... logging to /home/kochigami/.ros/log/1fbdeb72-b993-11ee-89c2-91b4da16ac22/roslaunch-kochigami-ThinkPad-P16-Gen-1-7460.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.232.44.17:44533/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [7491]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1fbdeb72-b993-11ee-89c2-91b4da16ac22
process[rosout-1]: started with pid [7524]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [7527]
disconnected
================================================================================REQUIRED process [naoqi_driver-2] has died!
process has finished cleanly
log file: /home/kochigami/.ros/log/1fbdeb72-b993-11ee-89c2-91b4da16ac22/naoqi_driver-2*.log
Initiating shutdown!
================================================================================
[naoqi_driver-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

3. When I changed 
`--qi-url=tcp://$(arg nao_ip):$(arg nao_port)` to `--qi-url=tcps://$(arg nao_ip):$(arg nao_port)`
in https://github.com/ros-naoqi/naoqi_driver/blob/master/launch/naoqi_driver.launch#L11 ,
I got the same results.

victorpaleologue, 2024-01-23:
How about: `roslaunch naoqi_driver naoqi_driver.launch naoqi_driver.launch nao_ip:=tcps://<ip> nao_port:=9443 network_interface:=<interface> password:=<password>`

kochigami, 2024-01-24:
Thank you very much for your advice!

```
roslaunch naoqi_driver naoqi_driver.launch nao_ip:=tcps://10.232.44.27 nao_port:=9443 network_interface:=<interface> password:=<password> 
```
I got `Bad address` error...
I need to find the code where handles the url address in naoqi_driver package.
```
... logging to /home/kochigami/.ros/log/22103430-ba6e-11ee-b5be-513faa04ecd2/roslaunch-kochigami-ThinkPad-P16-Gen-1-4340.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.232.44.17:33733/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [4373]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 22103430-ba6e-11ee-b5be-513faa04ecd2
process[rosout-1]: started with pid [4406]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [4409]
[W] 1706069268.256985 4409 qi.url: Could not parse port '//10.232.44.27:9443' from url 'tcp://tcps://10.232.44.27:9443' (errno:0, strerror:'Success')
[W] 1706069268.260886 4409 qi.url: Could not parse port '//10.232.44.27:9443' from url 'tcps://tcps://10.232.44.27:9443' (errno:0, strerror:'Success')
Connect error: 14: Bad address
================================================================================REQUIRED process [naoqi_driver-2] has died!
process has finished cleanly
log file: /home/kochigami/.ros/log/22103430-ba6e-11ee-b5be-513faa04ecd2/naoqi_driver-2*.log
Initiating shutdown!
================================================================================
[naoqi_driver-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

victorpaleologue, 2024-01-24:
And how about: `roslaunch naoqi_driver naoqi_driver.launch naoqi_driver.launch nao_ip:=<ip> nao_port:=9443 network_interface:=<interface> password:=<password>` ?
Sorry for going back and forth.

kochigami, 2024-01-25:
Thank you very much!! I'm also sorry for asking you again and again.

```
roslaunch naoqi_driver naoqi_driver.launch nao_ip:=192.168.8.100 nao_port:=9443  network_interface:=<interface> password:=<password>
```
I got `what():  Cannot find service 'ALMemory' in index`.
I think this error relates to this line: https://github.com/ros-naoqi/naoqi_driver/blob/master/src/helpers/driver_helpers.cpp#L48

```
... logging to /home/kochigami/.ros/log/5bec032c-bb5a-11ee-8498-1de1868ced84/roslaunch-kochigami-ThinkPad-P16-Gen-1-4204.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.8.162:33169/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [4238]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5bec032c-bb5a-11ee-8498-1de1868ced84
process[rosout-1]: started with pid [4271]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [4274]
Receiving information about robot model
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  Cannot find service 'ALMemory' in index
================================================================================REQUIRED process [naoqi_driver-2] has died!
process has died [pid 4274, exit code -6, cmd /home/kochigami/catkin_ws/devel/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://192.168.8.100:9443 --nao_ip=192.168.8.100 --nao_port=9443 --user=nao --password=<password> --roscore_ip=127.0.0.1 --network_interface=<interface> --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/kochigami/.ros/log/5bec032c-bb5a-11ee-8498-1de1868ced84/naoqi_driver-2.log].
log file: /home/kochigami/.ros/log/5bec032c-bb5a-11ee-8498-1de1868ced84/naoqi_driver-2*.log
Initiating shutdown!
================================================================================
[naoqi_driver-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

victorpaleologue, 2024-01-25:
Would you please try with the port 9503 instead?
Sorry for tiptoeing. I'll update the README when we're done.

kochigami, 2024-01-25:
It worked!!!!! Thank you very much for your help.
```
roslaunch naoqi_driver naoqi_driver.launch network_interface:=<interface> nao_ip:=192.168.8.100 nao_port:=9503 password:=<nao>
```
```
... logging to /home/kochigami/.ros/log/939b83c2-bb5c-11ee-8498-1de1868ced84/roslaunch-kochigami-ThinkPad-P16-Gen-1-6441.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.8.162:43901/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

auto-starting new master
process[master]: started with pid [6472]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 939b83c2-bb5c-11ee-8498-1de1868ced84
process[rosout-1]: started with pid [6505]
started core service [/rosout]
process[naoqi_driver-2]: started with pid [6508]
Receiving information about robot model
Robot detected/NAOqi version: Pepper 1.8A / 2.9.3.114
set prefix successfully to naoqi_driver
using ip address: 127.0.0.1 @ <interface>
found a catkin prefix /home/kochigami/catkin_ws/src/naoqi_driver/share/boot_config.json
load boot config from /home/kochigami/catkin_ws/src/naoqi_driver/share/boot_config.json
found a catkin URDF /home/kochigami/catkin_ws/src/naoqi_driver/share/urdf/pepper.urdf
RightBumperPressed
LeftBumperPressed
BackBumperPressed
ROS-Driver-RightBumperPressed : Start
HandRightBackTouched
HandRightLeftTouched
HandRightRightTouched
HandLeftBackTouched
HandLeftLeftTouched
HandLeftRightTouched
ROS-Driver-HandRightBackTouched : Start
FrontTactilTouched
MiddleTactilTouched
RearTactilTouched
ROS-Driver-FrontTactilTouched : Start
registered subscriber:	teleop
registered subscriber:	moveto
registered subscriber:	speech
nodehandle reset 
using master ip: http://127.0.0.1:11311
NOT going to re-register the converters
camera/bottom/image_raw is resetting
camera/bottom/image_raw reset
camera/depth/image_raw is resetting
camera/depth/image_raw reset
/diagnostics is resetting
/diagnostics reset
camera/front/image_raw is resetting
camera/front/image_raw reset
imu/base is resetting
imu/base reset
imu/torso is resetting
imu/torso reset
info is resetting
load robot description from file
info reset
camera/ir/image_raw is resetting
camera/ir/image_raw reset
/joint_states is resetting
/joint_states reset
laser is resetting
laser reset
odom is resetting
odom reset
sonar is resetting
sonar reset
teleop is resetting
teleop reset
moveto is resetting
moveto reset
speech is resetting
speech reset
get_robot_config is resetting
get_robot_config reset
set_language is resetting
set_language reset
get_language is resetting
get_language reset
naoqi_driver initialized
```

kochigami, 2024-01-25:
Should this port be 9503 as default?
If so, I will send a PR to update README or update `naoqi_driver.launch` :+1:

victorpaleologue, 2024-01-25:
Normally, if you provide a password it means you are connecting to a 2.9.x, so it tries 9503 by default. I see this is what you did, but the fallback did not happen, so you had to give it explicitly. It's good if the README recommends to put the port 9503 explicitly when it's a 2.8.x or higher.

kochigami, 2024-01-29:
Sorry for my late reply. 
I sent a PR: https://github.com/ros-naoqi/naoqi_driver/pull/164
If there any advice, please let me know.

I'll close this issue. Thank you very much again for your help!

closed 2024-01-29 by kochigami