ros-naoqi/naoqi_driver #163: Cannot compile for ROS Noetic Ubuntu 20.04 in WSL2

Nielsencu, 2024-01-27:
I followed the exact steps listed in http://wiki.ros.org/pepper/Tutorials for ROS Noetic and above.

However, I get the same error as this https://github.com/ros-naoqi/naoqi_driver/issues/114.

Where it complains that the LIBQI_VERSION is not defined.

Has anyone encountered the same issue?

victorpaleologue, 2024-01-29:
Are you sure you did not get an error at this line?
`rosdep install -i -y --from-paths ./naoqi_driver`
Could you share the output of the command?
Which OS are you running?

Nielsencu, 2024-01-29:
I'm running Ubuntu 20.04.

The output of the command is 
#All required rosdeps installed succesfully

The command rosdep update also shows
Add distro "noetic"
And no error can be seen

Is there a packaged version of the repository that can be installed via apt?

victorpaleologue, 2024-01-29:
Yes.
https://index.ros.org/p/naoqi_driver/#noetic
IIRC it's something like `apt install ros-noetic-naoqi-driver`.

victorpaleologue, 2024-01-29:
I've just updated the [README](https://github.com/ros-naoqi/naoqi_driver/blob/master/README.md) accordingly. Let me know if that's not enough info.

Nielsencu, 2024-01-29:
Thank you! Installation works fine.

One thing though, running roslaunch naoqi_driver naoqi_driver.launch gives 

ERROR: cannot launch node of type [naoqi_driver/naoqi_driver_node]: Cannot locate node of type [naoqi_driver_node] in package [naoqi_driver]. Make sure file exists in package path and permission is set to executable (chmod +x)

It looks like the naoqi_driver_node is missing.

Nielsencu, 2024-01-31:
Hi @victorpaleologue, can I check if this is indeed an issue with the packaged version of Noetic?

victorpaleologue, 2024-01-31:
Yes. I wanted to go check the contents of that ubuntu packages to try and figure out whether the node really is missing or not, or if something else is wrong. I'm happy if you can help debugging that, but it's matter of time before I'm on it.

Nielsencu, 2024-01-31:
Running rospack find naoqi_driver and cd-ing to the directory, and ls gives,
cmake (directory), launch (directory), package.xml, share (directory). There is no naoqi_driver_node to be found in any of these directories.

Nielsencu, 2024-02-10:
Hi @victorpaleologue if the node is indeed missing, may I check how can this be rectified?