ros-naoqi/naoqi_driver #165: error:  Session is already connected

UdayRockzz, 2024-02-11:
Hi, 

I am trying to connect the pepper robot with Ros. and I have installed all the required drivers for it. 

$ `roslaunch pepper_bringup pepper_full.launch nao_ip:=10.55.*.* network_interface:=wlxa842a191200*
` I used this command for connecting the pepper and I got the following error. 
[image]

How can we solve this error in ros-kinetic?

TIA

tungngovn, 2024-02-16:
I got the same error with you. Anyone has solution?

UdayRockzz, 2024-02-16:
downgrade the naoqi_driver to [5.11.](https://github.com/ros-naoqi/naoqi_driver/tree/0.5.11).  its worked for me.

ShifeiDuan, 2024-11-20:
I got the same error, after downgrade the naoqi_driver version to 0.5.11, I got the boost error.